Ultimaker/Cura #834: why it print the layer at the wrong place

snuchitp, 2016-07-01:
I used cura v 15.04.6 to display my model. From layer 1-9, it looked correct that built each layer up on top. But from layer 10- the end  started to build from the bottom of the model. The detailed lines are different too.

layer 0-9
[image]

layer 10 up
[image]

Thanks

thopiekar, 2016-07-01:
This is an "old" version of Cura.

Please reopen at: https://github.com/daid/Cura

closed 2016-07-01 by thopiekar

snuchitp, 2016-07-01:
I need to use this version because of my  3D printer

fieldOfView, 2016-07-01:
Feel free to use that version, but that version is not supported here. It still gets some support over at https://github.com/daid/Cura

thopiekar, 2016-07-01:
@snuchitp If your printer does not work at the moment with Cura 2.1, you can try to convert your printers definitions from "old" Cura to our new version. I'm sure someone can help getting it done if you open a new issue on that :wink:

snuchitp, 2016-07-01:
Could you please give me any suggestion how to convert my printers definitions from "old" Cura to your new version?

I am very very new user.
I have started to learn cura two days ago.

Thank you so much

thopiekar, 2016-07-01:
Welcome to Cura :smile: 

Which printer do you have?

Ghostkeeper, 2016-07-03:
Regardless of Cura 2, I suspect that what happens is that the new layers actually do get built on top of the old ones, but since the camera moves along with the highest layer, it looks like there are new layers on the bottom when you scroll up.