Ultimaker/Cura #837: STL file not displayed in Cura 2.1.2 OS X / Win

cekuhnen, 2016-07-01:
Slicer shows in layer view some layers but the UI in general does not show any imported data or material used. Fails in OS X and Win builds.

[image]
[Cable Cover 1.39.46 PM.stl.zip](https://github.com/Ultimaker/Cura/files/343945/Cable.Cover.1.39.46.PM.stl.zip)

thopiekar, 2016-07-01:
Can confirm this problem with Cura built from master:

```
file:///opt/cura/lib/qml/QtQuick/Controls/Private/MenuContentItem.qml:178: ReferenceError: Acccessible is not defined
file:///opt/cura/lib/qml/QtQuick/Controls/Private/MenuContentItem.qml:178: ReferenceError: Acccessible is not defined
file:///opt/cura/lib/qml/QtQuick/Controls/Private/MenuContentItem.qml:178: ReferenceError: Acccessible is not defined
file:///opt/cura/lib/qml/QtQuick/Controls/Private/MenuContentItem.qml:178: ReferenceError: Acccessible is not defined
2016-07-01 20:43:29,756 - DEBUG - /opt/cura/lib/cura/plugins/ImageReader/ImageReaderUI.py (_createConfigUI [81]): Creating ImageReader config UI
file:///opt/cura/lib/qml/QtQuick/Controls/Private/MenuContentItem.qml:178: ReferenceError: Acccessible is not defined
file:///opt/cura/lib/qml/QtQuick/Controls/Private/MenuContentItem.qml:178: ReferenceError: Acccessible is not defined
2016-07-01 20:43:40,244 - DEBUG - .../UM/Mesh/MeshData.py (calculateNormalsFromVertices [317]): Calculating normals took 0.004959583282470703 seconds
2016-07-01 20:43:40,245 - DEBUG - /opt/cura/lib/uranium/plugins/FileHandlers/STLReader/STLReader.py (read [22]): Loaded a mesh with 52224 vertices
QH6013 qhull input error: input is less than 3-dimensional since it has the same x coordinate

While executing:  | qhull i Qt
Options selected for Qhull 2012.1 2012/02/18:
  run-id 866404193  incidence  Qtriangulate  _pre-merge  _zero-centrum
  _max-width  0  Error-roundoff  0  _one-merge  0  _near-inside  0
  Visible-distance  0  U-coplanar-distance  0  Width-outside  0  _wide-facet  0
2016-07-01 20:43:40,263 - ERROR - .../UM/Logger.py (logException [49]): Exception: Exception in mesh loader
2016-07-01 20:43:40,291 - ERROR - .../UM/Logger.py (logException [49]): Traceback (most recent call last):
2016-07-01 20:43:40,292 - ERROR - .../UM/Logger.py (logException [49]):   File "/opt/cura/lib/python3/dist-packages/UM/Mesh/ReadMeshJob.py", line 55, in run
2016-07-01 20:43:40,292 - ERROR - .../UM/Logger.py (logException [49]):     node = self._handler.readerRead(reader, self._filename)
QH6013 2016-07-01 20:43:40,294 - ERROR - .../UM/Logger.py (logException [49]):   File "/opt/cura/lib/python3/dist-packages/UM/Mesh/MeshFileHandler.py", line 51, in readerRead
2016-07-01 20:43:40,294 - ERROR - .../UM/Logger.py (logException [49]):     extents = result.getMeshData().getExtents()
2016-07-01 20:43:40,294 - ERROR - .../UM/Logger.py (logException [49]):   File "/opt/cura/lib/python3/dist-packages/UM/Mesh/MeshData.py", line 153, in getExtents
qhull input error: input is less than 3-dimensional since it has the same x coordinate

While executing:  | qhull i Qt
Options selected for Qhull 2012.1 2012/02/18:
  run-id 866404193  incidence  Qtriangulate  _pre-merge  _zero-centrum
  _max-width  0  Error-roundoff  0  _one-merge  0  _near-inside  0
  Visible-distance  0  U-coplanar-distance  0  Width-outside  0  _wide-facet  0
2016-07-01 20:43:40,295 - ERROR - .../UM/Logger.py (logException [49]):     data = numpy.pad(self.getConvexHullVertices(), ((0, 0), (0, 1)), "constant", constant_values=(0.0, 1.0))
2016-07-01 20:43:40,296 - ERROR - .../UM/Logger.py (logException [49]):   File "/opt/cura/lib/python3/dist-packages/UM/Mesh/MeshData.py", line 228, in getConvexHullVertices
2016-07-01 20:43:40,299 - ERROR - .../UM/Logger.py (logException [49]):     convex_hull = self.getConvexHull()
2016-07-01 20:43:40,299 - ERROR - .../UM/Logger.py (logException [49]):   File "/opt/cura/lib/python3/dist-packages/UM/Mesh/MeshData.py", line 220, in getConvexHull
2016-07-01 20:43:40,299 - ERROR - .../UM/Logger.py (logException [49]):     self._computeConvexHull()
2016-07-01 20:43:40,300 - ERROR - .../UM/Logger.py (logException [49]):   File "/opt/cura/lib/python3/dist-packages/UM/Mesh/MeshData.py", line 212, in _computeConvexHull
2016-07-01 20:43:40,300 - ERROR - .../UM/Logger.py (logException [49]):     self._convex_hull = approximateConvexHull(points, MAXIMUM_HULL_VERTICES_COUNT)
2016-07-01 20:43:40,300 - ERROR - .../UM/Logger.py (logException [49]):   File "/opt/cura/lib/python3/dist-packages/UM/Mesh/MeshData.py", line 298, in approximateConvexHull
2016-07-01 20:43:40,301 - ERROR - .../UM/Logger.py (logException [49]):     hull_result = scipy.spatial.ConvexHull(vertex_data)
2016-07-01 20:43:40,301 - ERROR - .../UM/Logger.py (logException [49]):   File "scipy/spatial/qhull.pyx", line 2266, in scipy.spatial.qhull.ConvexHull.__init__ (scipy/spatial/qhull.c:21663)
2016-07-01 20:43:40,301 - ERROR - .../UM/Logger.py (logException [49]):   File "scipy/spatial/qhull.pyx", line 338, in scipy.spatial.qhull._Qhull.__init__ (scipy/spatial/qhull.c:3879)
2016-07-01 20:43:40,298 - CRITICAL - .../cura/CrashHandler.py (show [24]): An uncaught exception has occurred!
2016-07-01 20:43:40,302 - ERROR - .../UM/Logger.py (logException [49]): scipy.spatial.qhull.QhullError: Qhull error
2016-07-01 20:43:40,306 - CRITICAL - .../cura/CrashHandler.py (show [24]): Traceback (most recent call last):
2016-07-01 20:43:40,307 - CRITICAL - .../cura/CrashHandler.py (show [24]):   File "/opt/cura/lib/python3/dist-packages/cura/CuraApplication.py", line 406, in event
2016-07-01 20:43:40,307 - CRITICAL - .../cura/CrashHandler.py (show [24]):     return super().event(event)
2016-07-01 20:43:40,307 - CRITICAL - .../cura/CrashHandler.py (show [24]):   File "/opt/cura/lib/python3/dist-packages/UM/Qt/QtApplication.py", line 176, in event
[... cut ...]
2016-07-01 20:43:40,315 - CRITICAL - .../cura/CrashHandler.py (show [24]):   File "scipy/spatial/qhull.pyx", line 2266, in scipy.spatial.qhull.ConvexHull.__init__ (scipy/spatial/qhull.c:21663)
2016-07-01 20:43:40,315 - CRITICAL - .../cura/CrashHandler.py (show [24]):   File "scipy/spatial/qhull.pyx", line 338, in scipy.spatial.qhull._Qhull.__init__ (scipy/spatial/qhull.c:3879)
2016-07-01 20:43:40,316 - CRITICAL - .../cura/CrashHandler.py (show [24]): scipy.spatial.qhull.QhullError: Qhull error
```

thopiekar, 2016-07-01:
@cekuhnen If possible try another file format than .stl.
Would be useful for the others to find the reason of this.

sedwards2009, 2016-07-02:
@cekuhnen  Can we have the STL model? It looks like there is something degenerate in the geometry which the convex hull code doesn't like.

thopiekar, 2016-07-02:
Simon, the STL is in the zip file.

sedwards2009, 2016-07-02:
cool. I didn't notice it under the screenshots.

Ghostkeeper, 2016-07-03:
The problem seems to be that the mesh is extremely small, probably in units of metres instead of millimetres. A tiny shadow appears in the middle, but you can't zoom far enough in that the mesh even becomes a single pixel, so you can't select the mesh and you can't scale it up. I scaled it up in a different application by a factor of 1000 and it became a ring of about a centimetre wide.

So the problem at hand is that the auto-scaling doesn't trigger.

fieldOfView, 2016-07-03:
The problem lies before the auto-scaling-check; the model is so small that qhull seems to think it has 0 dimensions (ie: all points lie on the same coordinates). Looks like something's going wrong with rounding.

ChrisTerBeke, 2017-11-14:
Is this still an issue?

Ghostkeeper, 2017-11-22:
No, the model now gets scaled to 1000000% of original size (which is 10 times too big but we don't know the original unit).

closed 2017-11-22 by Ghostkeeper