Ultimaker/Cura #839: After firmware update not able to insert material

consp51, 2016-07-02:
After firmware update, I am not able to insert material, it will stuck at heating printhead for material insertion screen for long time. 
firmware version Vers: _2.1ex
Is there anyway i can upload the old version to fix this?

fieldOfView, 2016-07-02:
Strange, we have not had more reports of this. Does the printhead heat up at all? If you touch the printhead with a piece of filament, does it melt (I would not suggest touching it with your fingers to check the temperature)

Cura 2.1 does not currently let you select a firmware file. You could download 15.04.06 (or indeed an older version) and "update" the firmware to the version that comes with that.

Ghostkeeper, 2016-07-03:
The firmware update actually doesn't check if the update is newer than the version currently installed on the machine. So if you download any older version of Cura, it will install the firmware version that came along with that version.

consp51, 2016-07-05:
I think i figured out what's wrong, i select ultimaker  2 extended +
instead of ultimaker 2 extended。

Ghostkeeper, 2016-07-05:
Ah, yes. It assumes that the selected printer is also the connected one. This is a shortcoming of the current implementation (known bug): it doesn't detect which printer is connected. But it's hard to make that sort of thing work right for any printer, even just the ones we provide firmware for.

fieldOfView, 2016-07-05:
This should be much better in 2.2, now that firmware uploading is a machine action. Uploading firmware is available through the Manage Printers page.

thopiekar, 2016-07-05:
Well, I would still add a check here. It is possible to ask the printer for it's model name, if I'm not wrong. respecting that printers can be upgraded, I would add a dialog, which asks "Are you really sure about flashing firmware of UM2 Ex. Plus to UM2 Ex.?".

fieldOfView, 2016-07-05:
I thought it was impossible to query the firmware version over USB, but it seems to be very possible to get at least some data using the M115 command:
```
Send: M115
Recv: FIRMWARE_NAME:Marlin Ultimaker2; Sprinter/grbl mashup for gen6 FIRMWARE_URL:http://github.com/Ultimaker PROTOCOL_VERSION:1.0 MACHINE_TYPE:Ultimaker EXTRUDER_COUNT:1
Recv: ok
```
Notably you don't get the exact firmware version, but you do get the printer name and interestingly an extruder_count. Would be interested to know what other printers return.

thopiekar, 2016-07-05:
Looks like we don't need reinvent the wheel :smile: 

I think it would be also great if the maintainers of firmware take care about calling all printers there correctly :wink:

fieldOfView, 2016-07-05:
What does your BQ say?

thopiekar, 2016-07-05:
Moment..

thopiekar, 2016-07-05:
```
FIRMWARE_NAME:Marlin
FIRMWARE_VERSION:2.2.0
SOURCE_CODE_URL:http%3A//github.com/bq/Marlin
PROTOCOL_VERSION:1.0
MACHINE_TYPE:Hephestos_2
EXTRUDER_COUNT:1 X-FIRMWARE_LANGUAGE: X-BUILD_VERSION:"#475"
X-SERIAL_NUM:<FOO-BAR>
ok
```

thopiekar, 2016-07-05:
I would even use this for USB connection testing then asking for the extruder temp.

Btw. we could use that M115 info to discover printers for example.

fieldOfView, 2016-07-05:
M408 would have been nice too, but Marlin does not support it:
http://reprap.org/wiki/G-code#M408:_Report_JSON-style_response

Discovery would be tricky to maintain with different firmwares from different manufacturers; just look st the different fields reported by BQ and UM. Let alone what happens with printers that have "custom" firmware.

thopiekar, 2016-07-05:
Well, let's advertise on that and make it work on the next firmware updates(?).
On unknown M115 responses I would warn before flashing and for custom firmware (eg. when Prusa i3 selected) I would give the possibility to select the .hex.

Well, this could be done by:
```
if "RepRapFirmware" in GCodeResponse["M115"]["FIRMWARE_NAME"]: GCodeRequest("M408")
```

However, I'm planning to report an JIRA ticket for something else regarding this :wink:

thopiekar, 2016-07-06:
Maybe I should try the "vanilla" firmware using Marlin's examples.
https://github.com/MarlinFirmware/Marlin/tree/RC/Marlin/example_configurations

At least I could get a rid of this M800 custom G-Code for bed leveling.

ianpaschal, 2018-02-27:
Hasn't been touched in a year and a half.

Ghostkeeper, 2018-03-16:
That doesn't make it any less of an issue. We get reports from Ultimaker's forum where people bricked* their printer because they had a different printer type in Cura than their actual printer model.

The M115 command is a good solution that I didn't know of. Let's propose that.

* Most of the time it's not really bricked as long as the PCB of the printer is supported by that firmware enough that USB connections still work; because then you can still upload a new firmware via USB.

ianpaschal, 2018-03-16:
In that case we should be either marking the newer issues duplicates referring back to here, or just continue in the new ones.

But leaving stuff open indefinitely is not really logical. I'm sure there's still technically issues with hooking up certain VCRs but you gotta set a timeout on these kinds of things. With all support things when stuff gets too old you have to drop it at some point.

In this case we had no one commenting on it, re-reporting it, or working on it for a year and a half. That's deeeeaaaaad.

closed 2018-11-26 by Appesteijn