Ultimaker/Cura #840: Cura uncaught exception

cmarciales, 2016-07-03:
Version: 15.06.03
Platform: Windows-7-6.1.7601-SP1
Qt: 5.4.1
PyQt: 5.4.1

Exception:
Traceback (most recent call last):
  File "cura_app.py", line 9, in <module>
  File "c:\dev\uranium\UM\Application.py", line 329, in getInstance
  File "c:\dev\cura\cura\CuraApplication.py", line 56, in __init__
  File "c:\dev\uranium\UM\Qt\QtApplication.py", line 48, in __init__
  File "c:\dev\uranium\UM\Application.py", line 67, in __init__
  File "c:\dev\uranium\UM\Preferences.py", line 68, in readFromFile
  File "c:\dev\python34\lib\_collections_abc.py", line 504, in __iter__
  File "c:\dev\python34\lib\configparser.py", line 1204, in __getitem__
  File "c:\dev\python34\lib\configparser.py", line 773, in get
  File "c:\dev\python34\lib\configparser.py", line 374, in before_get
  File "c:\dev\python34\lib\configparser.py", line 423, in _interpolate_some
configparser.InterpolationSyntaxError: '%' must be followed by '%' or '(', found: "%20Workspace/Poo%20bag%20dispenser\\xff\\xff\\xff\\xff\\x00\\x00\\x00\\x01'"

fieldOfView, 2016-07-03:
Don't use 15.06.03; it is more than a year old and much has been fixed since then. Please update to 2.1.2, available from http://software.ultimaker.com

closed 2016-07-03 by fieldOfView

cmarciales, 2016-07-03:
I just realized this. It seems that installing 2.1.2 did not update my
shortcuts, so double clicking on an stl file called the old program. Sorry
for the bother, and thanks for letting me know.