Ultimaker/Cura #845: I can't find delta machines

famfilho, 2016-07-03:
Hi I'm from Brazil and I have a delta printer that I bought here

I downloaded the Cura 2.1.2 and for my surprise I can't find any delta machines. So, how can I do to setting my machine in this SW version?
data of my machine:
Company: PUZZLES 3D (this is a very small campany)
Bed Diameter: 200mm
Height: 300mm
Nozzle diameter: 0,4mm
Extruder N°: single
Control Board: RAMPS 1.4
[image]

fieldOfView, 2016-07-04:
Cura 2.x does not have support for circular beds at the moment. It has to be re-implemented, either by us or by the community.

Cura 15.04 is still available for download if you want support for delta printers out of the box.

closed 2016-07-04 by nallath

famfilho, 2016-07-04:
Dear all
Thank you for the quickly return.
Sincerely.
Francisco Machado 

Enviado por TypeApp