Ultimaker/Cura #849: 2 slicing profiles selected at the same time

MarcusWolschon, 2016-07-06:
Cura 2.1.2 on MacOS.
I just added the file 
https://gist.github.com/fieldOfView/ae85f755fc70d1c65e6968a870dfef0c
to ~/.cura/machines
https://github.com/Ultimaker/Cura/issues/847#issuecomment-230454073
then created a machine with that new type and selected an existing profile.

I'm sure this is not supposed to happen:
[image]

MarcusWolschon, 2016-07-06:
I got the same thing when adding a new machine.
Goes away after restarting Cura.

Then I can select another slicing profile but the selection shown in the collapsed list-view stays the same regardless of what I select in the expanded list.

Ghostkeeper, 2016-07-12:
Fixed in version 2.2.

closed 2016-07-12 by Ghostkeeper