Ultimaker/Cura #850: Sliced file not saved when custom machine is used in alternative json location

MarcusWolschon, 2016-07-06:
Cura 2.1.2 on MacOS.
I just added the file
https://gist.github.com/fieldOfView/ae85f755fc70d1c65e6968a870dfef0c
to ~/.cura/machines
as per https://github.com/Ultimaker/Cura/issues/847#issuecomment-230454073
then created a machine with that new type and selected an existing profile.

Slicing works fine, I can select a location to save the .gcode file, but no file is actually saved there.

MarcusWolschon, 2016-07-06:
When you copy fdmprinter.json, ultimaker2.json ... from
..../Cura 2.1.2.app/Contents/Resources/cura/resources/machines
into 
~/.cura/machines

then everything works.
Aparently it looks for the parent files only in the same directory, where it got the json file given by the UI.

awhiemstra, 2016-07-06:
Yeah this is a known issue with the engine and how things are communicated between the two. For 2.2 this will be less of an issue.

ChrisTerBeke, 2017-11-14:
Is this still an issue?

Ghostkeeper, 2017-11-22:
If I understand this correctly, it was worked around in 2.2 and completely resolved by https://github.com/Ultimaker/Uranium/commit/a3c77e86fc23c91d5084a7530616ead95776f86e in 3.0.

closed 2017-11-22 by Ghostkeeper