Ultimaker/Cura #851: Cura engine wont slice or work properly version 2.1.2

prusanoff, 2016-07-06:
Hey guys Cura is acting crazy on my mac 
    I ran it as Supper User "sudo" here is what im getting ...

`
 - DEBUG - Calculating normals took 0.047377824783325195 seconds
 - DEBUG - Loaded a mesh with 862464 vertices
 - DEBUG - Loading mesh took 3.874324083328247 seconds
 - DEBUG - Attempting to kill the engine process
 - DEBUG - Killing engine process
 - DEBUG - Engine process is killed. Recieved return code -15
 - INFO - Backend crashed or closed. Restarting...
 - DEBUG - Attempting to kill the engine process
 - INFO - Started engine process: /Applications/Cura.app/Contents/MacOS/CuraEngine
 - DEBUG - Backend connected on port 49674
 - DEBUG - Autosaving preferences, instances and profiles
 - DEBUG - Backend quit with return code 0. Resetting process and socket.
 - INFO - Started engine process: /Applications/Cura.app/Contents/MacOS/CuraEngine
 - DEBUG - Backend connected on port 49674
 - DEBUG - Setting active view to SolidView
 - DEBUG - Setting active view to LayerView
 - ERROR - There are no file types available to write with!`

nallath, 2016-07-06:
Have you als tried saving using file->save all?

Ghostkeeper, 2016-07-07:
I suspect there might be some problem in the machine definition you're using. The machine definition specifies which file type is being used to output g-code.

closed 2016-12-26 by thopiekar