Ultimaker/Cura #853: Profile management - update profile button

danilius, 2016-07-06:
I'm finding the profile management a bit fiddly.

I would like to see an "Update profile" button, as well as a "Duplicate Profile" button.

nallath, 2016-07-06:
2.2 (The stuff that is on master now) has a duplicate & update profile function.

closed 2016-07-06 by nallath

fieldOfView, 2016-07-06:
> 2.2 (The stuff that is on master now) has a duplicate & update profile function. 

...and so much more...

@danilius: I totally agree with this request. I feel partly responsible for this, because I implemented the profile management. Rest assured that it has been reimplemented by people much smarter than me (and with more time than I had).

danilius, 2016-07-06:
Thanks for that.

I would love to download the build from git, but had too many problems with
getting it to compile or run so I gave it up. It would be a great idea to
have nightly builds, which we can download, test and delete as we see fit!

nallath, 2016-07-07:
If you drop me a mail (or contact me on irc / forum) I can send you a nightly master build.