Ultimaker/Cura #855: Feature suggestion: z hop on top layer only

filipgoc, 2016-07-06:
z-hop is a common solution against top layer nozzle scars. BUT z hop slows down the print overall. 
let's have a setting to **only do z-hop on the top layer (or top n layers)**

I feel silly suggesting this because it feels like a no-brainer feature, but i see it nowhere and github search is silent, too. 

Thoughts?

Ghostkeeper, 2016-07-07:
Z-hop is also useful to prevent scars and blips on the walls.

In any case, to prevent scars on the top layer only, it would have to activate only in the top skin, which is not necessarily the top layer (for example, if you're printing a letter L, then there is lots of skin on the lower terrace where you'd want to hop as well).

filipgoc, 2016-07-07:
right you are @Ghostkeeper

closed 2018-02-27 by ianpaschal