Ultimaker/Cura #856: Reboots UM2+ upon startup

swp466, 2016-07-07:
When 2.1.2 is started, the connected UM2+ resets.  The same reset as when firmware is updated.  Imagine my surprise when this occurred the first time about halfway through a print.  I'll be moving back to 15.04.6.

nallath, 2016-07-08:
This is because how arduino based systems handle a connection request; They reset.

closed 2016-07-08 by nallath

fieldOfView, 2016-07-08:
I am sorry to hear about your print. You were printing from SD-Card, while having your printer connected with USB? The extremely unhelpful answer is "don't do that". 

Unfortunately the Arduino hardware and drivers are designed to reset the board when the serial port is opened. Any application trying to see what devices have a serial port is going to reset your printer, and I think rebooting your computer is going to have the same effect (but I am not sure about that).

If Cura wants to offer USB printing, it is going to have to see if a printer is connected through USB. You can not look for printers connected through USB without this happening. Our advice is to either print from SD or have your printer connected through USB, but not both.
 
(too slow)

nallath, 2016-07-08:
We might want to look into disabling the auto detection in preferences somewhere though. Shouldn't be too hard. Got any ideas on that fieldofview?

fieldOfView, 2016-07-08:
Then how do we decide whether or not to show the "Print via USB" button? If we fix the plugin-system to allow disabling plugins we could give users the option to disable "Print via USB" altogether, but that would also disable uploading firmware.

nallath, 2016-07-08:
In that case we simply don't show the button at all unless the user starts looking for a printer (or somehwat similar to how we want to handle octoprint; Only if you are "linked" you get a print with button) 

We could also make a "go to old cura way of handling this" button.

swp466, 2016-07-08:
A bit of clarification -- previous versions of Cura (15.04.4 and 15.04.6) do NOT exhibit this behavior, and neither does turning on, turning off or rebooting the computer.  The ONLY things that have caused this were: (1) installing Cura 2.1.2 and (2) running Cura 2.1.2.  This version of Cura is doing something very different than the previous version.  It doesn't even seem to be based on the earlier version as the entire look and feel are very different.

fieldOfView, 2016-07-08:
>  It doesn't even seem to be based on the earlier version as the entire look and feel are very different.

That is correct. The Cura frontend has been rewritten from scratch, and comparing the new Cura with the old Cura is like comparing two entirely different programs.

swp466, 2016-07-08:
That figures.  So is there any way to prevent the new Cura from rebooting the machine when it's started?  I know I can just unplug the USB, but that takes away from some of the convenience.  I guess that's the "official" answer then?

swp466, 2016-07-08:
Also, will the new version eventually support PlugIns?  I really liked the Tweak at Z.

fieldOfView, 2016-07-08:
Unplugging USB when you don't use it is the easiest (and safest!) way for now.

Have a look at the Extensions -> Postprocessing menu.

nallath, 2016-07-08:
The difference between Old Cura and New Cura in this regard is the following: 
Old cura -> Saw any kind of USB device that looks like an arduino and assumed it was a printer. 
New cura -> Sees any kind of USB device that looks like an arduino, tries to connect with it to figure out if it's a printer or not. If it's a printer, it adds it to the output devices.

The advantage of the new approach is that we now know if a device is actually a printer before we try to print. The obvious disadvantage is the resetting of the printer.

swp466, 2016-07-08:
I see.  I think the disadvantage may outweigh the advantage in this case.  Would it be feasible to add a dialog box upon startup that says something like "check for printer?" (with a warning that it will reset the printer), and maybe a checkbox to remember the decision (along with a manual selection in the settings if a user later changes their mind)?  Just a thought...

nallath, 2016-07-11:
We could add this as a preference option. We already did this for quite a number of behaviors that some people do like and others don't (Auto scale, focus camera on select, etc)

fieldOfView, 2016-07-11:
It's been a while since I used Cura for USB-printing (I use OctoPrint myself). Is it acceptable (and the current behavior in Cura 15.04) if the printer resets when starting a print?

nickthetait, 2016-07-11:
I'd say yes, that is fine if printer resets when starting a print.

Ghostkeeper, 2016-07-12:
When you start a print on a machine that is already printing you're basically asking for trouble, as far as I'm concerned. And when you start a print on a machine that's not already running it shouldn't matter that the machine reboots as long as it doesn't break the machine.

fieldOfView, 2016-07-22:
> When you start a print on a machine that is already printing 

That is something we can and should catch from our side.

vilemmarsik, 2018-06-15:
I consider this quite a problem - printing from Pronterface, sometimes using Cura for slicing that Slic3r can't do well. Cura resets my printer's coordinates when just started. So far was cautious enough not to start it in the middle of a print, I guess it would get ruined. Thinking about writing SELinux rules to prevent Cura from touching USB at all.

fieldOfView, 2018-06-15:
You can disable the USB Printer plugin (via Toolbox in 3.4, or the plugin browser in 3.3)

Vandrasc, 2018-06-15:
It is planned to be fixed (no date for it, yet), the ticket is currently in our backlog.

CURA-5406

vilemmarsik, 2018-06-15:
Thanks, disabling the USB plugin helped.