Ultimaker/Cura #860: Crash while moving objects around print area.

jjman6, 2016-07-09:
I get very frequent crashes while trying to locate/move objects around print area.  So far seems like it only occurs when there are more than 1 item on print area.   Doesn't happen 100% of the time.  but probably around 85%.  it'll crash 5-10times before i can successfully get all the items i want put on print area without it crashing.  

This is becoming a deal breaker and i will start using other slicers unless i can get this resolved.

I'm using Kubuntu 15.10 64bit w/ Nvidia 352.63 drivers.  
Cura 2.1.2
Let me know what other info i can provide.

Ghostkeeper, 2016-07-10:
We haven't seen this with our testing and QA, as far as I know.

I'd like to see if there's something in Cura's log. It's located in ~/.local/share/cura/cura.log. Could you paste a bit at the end of that log here? Specifically if it names any errors.

jjman6, 2016-07-11:
I think upon further testing its only crashing IF i'm moving parts while its re-slicing things.    
As its a rather annoying feature that every single action/change to settings causes it to automatically reslice everything on the fly (which slows everything down considerably).  It would be far more preferable to have it slice once u were done making changes & went to save or switching to layers.  Rather than having to wait for it to re-slice on EVERY single movement u make on platform And every setting change.  cause if its slicing while ur trying to still move things and change things it seems that's where the crash comes in, not to the mention the annoying slow down. as u have to wait to add parts & wait again to move them until its DONE slicing the last one in..  

Is there a way to disable that so its not constantly re-slicing while ur trying to adjust settings & move parts around?  

Here i was able to reproduce crash again this is the part of log while it crashed: 

Cura.log
....................
2016-07-11 11:26:26,535 - DEBUG - Shutting down cura
2016-07-11 11:26:28,138 - ERROR - Exception while emitting shutdown signal: NotImplementedError('Stop should be implemented by subclasses',)
2016-07-11 11:26:28,162 - DEBUG - Attempting to kill the engine process
2016-07-11 11:26:28,162 - DEBUG - Killing engine process
2016-07-11 11:26:28,163 - DEBUG - Engine process is killed. Recieved return code -15
2016-07-11 11:26:37,913 - INFO - Loaded plugin FileLogger
2016-07-11 11:26:37,949 - INFO - Loaded plugin TranslateTool
2016-07-11 11:26:37,976 - INFO - Loaded plugin ScaleTool
2016-07-11 11:26:37,993 - INFO - Loaded plugin RotateTool
2016-07-11 11:26:38,000 - INFO - Loaded plugin CameraTool
2016-07-11 11:26:38,010 - INFO - Loaded plugin MirrorTool
2016-07-11 11:26:38,029 - INFO - Loaded plugin SelectionTool
2016-07-11 11:26:38,029 - WARNING - Plugin ConsoleLogger was already loaded
2016-07-11 11:26:38,035 - INFO - Loaded plugin OBJWriter
2016-07-11 11:26:38,042 - INFO - Loaded plugin OBJReader
2016-07-11 11:26:38,087 - INFO - Loaded plugin 3MFWriter
2016-07-11 11:26:38,094 - INFO - Loaded plugin STLWriter
2016-07-11 11:26:38,100 - INFO - Loaded plugin STLReader
2016-07-11 11:26:38,111 - INFO - Loaded plugin LocalFileOutputDevice
2016-07-11 11:26:38,152 - INFO - Checking for new version of cura
2016-07-11 11:26:38,154 - INFO - Loaded plugin UpdateChecker
2016-07-11 11:26:38,306 - INFO - Loaded plugin SliceInfoPlugin
2016-07-11 11:26:38,394 - INFO - Loaded plugin ImageReader
2016-07-11 11:26:38,403 - INFO - Loaded plugin RemovableDriveOutputDevice
2016-07-11 11:26:38,409 - INFO - Loaded plugin CuraProfileWriter
2016-07-11 11:26:38,433 - INFO - Loaded plugin PostProcessingPlugin
2016-07-11 11:26:38,438 - INFO - Loaded plugin GCodeWriter
2016-07-11 11:26:38,445 - INFO - Loaded plugin PerObjectSettingsTool
2016-07-11 11:26:38,452 - INFO - Loaded plugin 3MFReader
2016-07-11 11:26:38,453 - WARNING - Plugin CuraEngineBackend was already loaded
2016-07-11 11:26:38,535 - INFO - Loaded plugin USBPrinting
2016-07-11 11:26:38,547 - INFO - Loaded plugin ChangeLogPlugin
2016-07-11 11:26:38,616 - INFO - Loaded plugin CuraProfileReader
2016-07-11 11:26:38,624 - INFO - Loaded plugin LayerView
2016-07-11 11:26:38,645 - INFO - Loaded plugin XRayView
2016-07-11 11:26:38,651 - INFO - Loaded plugin GCodeProfileReader
2016-07-11 11:26:38,657 - INFO - Loaded plugin AutoSave
2016-07-11 11:26:38,663 - INFO - Loaded plugin LegacyProfileReader
2016-07-11 11:26:38,669 - INFO - Loaded plugin SolidView
2016-07-11 11:26:38,681 - ERROR - Exception starting plugin Doodle3D: NotImplementedError('Start should be implemented by subclasses',)
2016-07-11 11:26:38,681 - INFO - Loaded plugin Doodle3D
2016-07-11 11:26:38,683 - INFO - Command line arguments: {'disable-textures': False, 'debug-mode': False, 'file': [], 'external-backend': False}
2016-07-11 11:26:38,855 - WARNING - No active machine found when trying to save setting visibility
2016-07-11 11:26:38,857 - DEBUG - Attempting to kill the engine process
2016-07-11 11:26:38,989 - INFO - Started engine process: /opt/cura/bin/CuraEngine
2016-07-11 11:26:39,048 - DEBUG - Backend connected on port 49674
2016-07-11 11:26:39,049 - DEBUG - Setting active view to SolidView
2016-07-11 11:26:39,269 - DEBUG - Calculating normals took 0.0009236335754394531 seconds
2016-07-11 11:26:39,269 - DEBUG - Loaded a mesh with 9228 vertices
2016-07-11 11:26:39,945 - DEBUG - Loaded theme /opt/cura/share/cura/resources/themes/cura
2016-07-11 11:26:40,854 - DEBUG - Initialized OpenGL subsystems.
2016-07-11 11:26:40,854 - DEBUG - OpenGL Version:  4.5.0 NVIDIA 352.63
2016-07-11 11:26:40,855 - DEBUG - OpenGL Vendor:   NVIDIA Corporation
2016-07-11 11:26:40,855 - DEBUG - OpenGL Renderer: GeForce GT 650M/PCIe/SSE2
2016-07-11 11:26:51,406 - DEBUG - Autosaving preferences, instances and profiles
2016-07-11 11:26:56,295 - DEBUG - Calculating normals took 0.003499269485473633 seconds
2016-07-11 11:26:56,295 - DEBUG - Loaded a mesh with 77316 vertices
2016-07-11 11:26:56,321 - DEBUG - Loading mesh took 0.4959902763366699 seconds
2016-07-11 11:26:57,001 - DEBUG - Attempting to kill the engine process
2016-07-11 11:26:57,001 - DEBUG - Killing engine process
2016-07-11 11:26:57,002 - DEBUG - Engine process is killed. Recieved return code -15
[... 81 lines cut ...]
2016-07-11 11:27:24,673 - DEBUG - Backend connected on port 49674
[... cut ...]
2016-07-11 11:27:38,164 - INFO - Backend crashed or closed. Restarting...
2016-07-11 11:27:38,267 - DEBUG - Attempting to kill the engine process
2016-07-11 11:27:38,278 - INFO - Started engine process: /opt/cura/bin/CuraEngine
2016-07-11 11:27:38,308 - DEBUG - Backend connected on port 49674
2016-07-11 11:27:42,224 - DEBUG - Attempting to kill the engine process
2016-07-11 11:27:42,224 - DEBUG - Killing engine process
2016-07-11 11:27:42,227 - DEBUG - Engine process is killed. Recieved return code -15

Ghostkeeper, 2016-07-12:
You could try this add-on to pause auto-slicing for you: https://github.com/fieldOfView/PauseBackendPlugin

Ghostkeeper, 2016-07-12:
Still can't reproduce your actual issue though :( But I have no access to your OS/hardware combination.

jjman6, 2016-07-13:
I'll give the add-on a try.  I haven't noticed any crashes since i've
discovered it seems to be linked to movement while its slicing.  So waiting
to move parts around seems to mitigate the issue so far.  Thanks for
looking into it.

fieldOfView, 2016-07-13:
What exactly is crashing? Cura (the GUI) or CuraEngine?

jjman6, 2016-07-13:
Cura itself closes.

Ghostkeeper, 2016-07-14:
Perhaps it goes out of memory? Can't think of much else that could evade our error logging.

jjman6, 2016-07-14:
maybe if there's a memory leak in Cura.
I've got 8gb of system RAM plus another 5gb of swap so it would have to be
a memory leak if so i would think.

I'm almost certain now that it only occurs while moving parts around when
its doing its auto-slice.  As i've paid closer attention to my moving parts
while not slicing and haven't gotten it to crash yet.

nallath, 2016-07-15:
You could test your theory by setting your layer height at 0 and moving objects.

Ghostkeeper, 2016-07-15:
Hmm, well perhaps it is a segfault/crash in libArcus or Protobuf then?

mokusolo, 2016-09-07:
I wonder what does"INFO - Backend crashed or closed. Restarting..." mean.Does it just because backend was closed?I found occasionally the restarting takes 3 second which may affect the next time slicing.(Just guess)
And I searched repository(include UM),still can't find where to log this sentence.Could you please tell me?Thanks!

awhiemstra, 2016-09-07:
It means that, for some reason, the backend process quit. This usually means the backend completed successfully if it exited with error code 0. If the error code is != 0, this usually means a crash in the backend.

Note that the logging message comes from plugins/CuraEngineBackend/CuraEngineBackend.py in the Cura repository.

mokusolo, 2016-09-08:
Thanks a lot!

ianpaschal, 2018-03-26:
This doesn't seem to happen in 3.2 anymore so I'm going to go ahead and close this.

closed 2018-03-26 by ianpaschal