Ultimaker/Cura #862: path of files

RoRi3D, 2016-07-10:
Today I noticed that Cura 2.1.2 is not using, when saving the G-code,  as default the path of the STL opened. 
Can this be fixed?
Now you have to click many times before you can save.

Ghostkeeper, 2016-07-10:
The expected behaviour is not that it should take the directory from where the STL was loaded, but that it should remember the directory you used last to save to.

nallath, 2016-07-11:
Nope it doesn't. It used to be that way but I was told to change this. The behavior now is that it _always_ looks in the home directory.

Ghostkeeper, 2016-07-12:
We have put this on the planning again to reinstate the remembering of save/load locations.

nallath, 2017-07-27:
I think this has already been fixed.

closed 2017-07-27 by nallath