Ultimaker/Cura #864: Start filament extrusion lenght

RoRi3D, 2016-07-10:
Previous Cura had the option to use the "start g-code"-field to indicate how much filament should be extruded before starting with the print.
Advise: either to add "start g-code" box again (like now is done with the post-processing)
or add it as option in the menu materials.

Ghostkeeper, 2016-07-10:
We still have the start g-code setting, but it is not exposed to the user directly. You can still modify it in the machine definition files, but not from the GUI.

Yet.

We still plan on re-introducing the machine editor. That would be a good place to modify the starting g-code.

Also know that one of the functions of printing a skirt is to make sure that the filament is properly flowing. Until we've re-introduced the machine editor, I advise to increase the Skirt Line Count setting a few ticks.

closed 2017-11-14 by ChrisTerBeke

Ghostkeeper, 2017-11-23:
The machine editor is now in the release.