Ultimaker/Cura #865: Cura does not start

Serj16, 2016-07-11:
Oops!
An uncaught exception has occurred!

Version: 15.06.03
Platform: Windows-8-6.2.9200
Qt: 5.4.1
PyQt: 5.4.1

Exception:
Traceback (most recent call last):
  File "cura_app.py", line 9, in <module>
  File "c:\dev\uranium\UM\Application.py", line 329, in getInstance
  File "c:\dev\cura\cura\CuraApplication.py", line 56, in __init__
  File "c:\dev\uranium\UM\Qt\QtApplication.py", line 48, in __init__
  File "c:\dev\uranium\UM\Application.py", line 67, in __init__
  File "c:\dev\uranium\UM\Preferences.py", line 68, in readFromFile
  File "c:\dev\python34\lib\_collections_abc.py", line 504, in __iter__
  File "c:\dev\python34\lib\configparser.py", line 1204, in __getitem__
  File "c:\dev\python34\lib\configparser.py", line 773, in get
  File "c:\dev\python34\lib\configparser.py", line 374, in before_get
  File "c:\dev\python34\lib\configparser.py", line 423, in _interpolate_some
configparser.InterpolationSyntaxError: '%' must be followed by '%' or '(', found: "%D0%A4%D1%80%D0%B5%D0%B7%D0%B5%D1%80%D0%BD%D1%8B%D0%B9%20%D1%81%D1%82%D0%B0%D0%BD%D0%BE%D0%BA%20%D1%81%20%D0%A7%D0%9F%D0%A3/3D%20Printer/1%20MODELS/models\\xff\\xff\\xff\\xff\\x00\\x00\\x00\\x01'"

fieldOfView, 2016-07-11:
Please use a newer version than 15.06.03, and please provide a bit more info when you report an issue.

closed 2016-07-11 by fieldOfView

Serj16, 2016-07-11:
version 15.04.06 OK

fieldOfView, 2016-07-11:
Version 2.1.2 is the current version

Serj16, 2016-07-12:
Installed 2.1.2. OK. I Malyan M 180. How to connect plugin x3g print from SD card?

Ghostkeeper, 2016-07-12:
There are various X3G plug-ins for Cura, but I think the most recent one is [this one](https://github.com/Ghostkeeper/X3GWriter) which was for version 15.10. I (or someone) needs to update that one, and perhaps inform @markwal once that's done since he's maintaining the most recent GPX version that I know of.

Serj16, 2016-07-13:
Thank you. How to connect for version 15.04 or 2.1.2 ?

Ghostkeeper, 2016-07-13:
Mankati has provided a plug-in that works for 15.04, if I recall correctly. For 2.1.2 nothing can make it work yet.

Serj16, 2016-07-13:
Please explain how to connect it to 15.04. I'm a beginner.

fieldOfView, 2016-07-13:
We don't support 15.04 here. Normally I would point you to https://github.com/daid/cura for 15.04, but given that this is about a plugin provided by Mankati you should probably contact them.

markwal, 2016-07-13:
@Serj16, while you're waiting for a plugin for a new version of Cura, you can just use GpxUi to convert from gcode to x3g: https://markwal.github.io/GpxUi