Ultimaker/Cura #868: Feature Request: Save/Load session as project-file

OctagonHex, 2016-07-11:
Hi,
I would like to have an option to save all the settings and the file name of 3D-file in a "project-like" file. So that, when I open that project again, my model, the orientation it is in, and all the print settings are restored.

This is useful, if you come back after some time and look at what you did when you printed this or that model, how this support setting/pattern/... worked better or not. It's simply not possible and/or practical to recover that from the GCode-files.

Thanks!

fieldOfView, 2016-07-11:
I do agree with the feature request, but more for placement and per-object-settings of multiple files.

nallath, 2016-07-11:
3mf should be able to support this without a whole lot of problems.

OctagonHex, 2016-07-11:
I tested 3MF-file format, but it doesn't work.

What I did:
- Open a model in Cura 2.1.2
- change layer thickness to e.g. 0.1234
- File-> save selection as file -> test.3MF
- Clear Build plate
- Reset settings to "Fast Print"
- Open the previously generated test.3MF
- layer thickness is _not_ restored to previous value.

nallath, 2016-07-11:
Because i said; "Should" not "Can" ;)

Ghostkeeper, 2016-07-12:
3MF is able to support it, but it's just not implemented in Cura yet :yum:

This one has been on the planning for a while now. Workspace saving (On Jira: CURA-1263).

nickthetait, 2016-07-13:
Would like to add a +1, I have had several users request this functionality. As a extra bonus I think this would be a very helpful feature for debugging and testing purposes.

Ghostkeeper, 2016-07-14:
Our overlords have put this up as feature for release 3.1. That'd take a while...

nallath, 2016-07-14:
Still think we should do it at an earlier data.

nallath, 2017-07-27:
This has already been implemented since now.

closed 2017-07-27 by nallath