Ultimaker/Cura #870: Cura 2.1 forgets build volume height

MarcusWolschon, 2016-07-11:
Cura 2.1.0, 2.1.1 and 2.1.2
It often forgets the height of the build volume of my Ultimaker II extended.
Then the maximum height is some kind of default.
See attached picture.
The only thing that helps is to remove all machine_instances and re-create the printer.
[image]

fieldOfView, 2016-07-11:
Set "print sequence" back to "all at once". For "one at a time" mode, the build volume height is limited by the distance between the bed and the gantry (axes); anything higher might collide with the gantry.

Cura 2.2 shows a message when activating "one at a time" printing.

closed 2016-07-11 by fieldOfView

MarcusWolschon, 2016-07-12:
We should change this into a feature-request then:

When loading a model that can not be printed in "one at a time" mode,
we should switch to "all at once" mode and warn the user instead if refusing to print the model at all.

nallath, 2016-07-12:
The switching was the behavior in old cura, but it was bloody annoying.

MarcusWolschon, 2016-07-12:
With all due respect, "annoying" is much better then "I can't slice at all and there is no indication why Cura is not working".

Also apparently for newly created printers this is ignored as I'm using the same, existing slicing profile with the same "print one at a time" setting on the newly created printer without any height restriction.

In the profile "one at a time" is enabled because that's what the slicer shall do IF POSSIBLE as it created much better parts.

nallath, 2016-07-12:
It was annoying because it would change what it was doing, without clear indication why it was doing that. This meant that you quite often found out hours later that something weird happend. The forum is littered with questions about this where this went wrong for people.

One at a time doesn't always create better parts. The UM robot for instance looks much better if you print 4-5 at a time, due to min layer time issues. 

That the one at a time isn't handled correcly for a new printer is a bug and should be adressed.

MarcusWolschon, 2016-07-12:
What about a visual indication for the reason, why the height is limited?

awhiemstra, 2016-07-12:
As mentioned by @fieldOfView:

> Cura 2.2 shows a message when activating "one at a time" printing.

fieldOfView, 2016-07-12:
Not only does it show the message when the user changes that one property, the message is also shown when a profile is selected that has that property set to One at a Time, and even when selecting a printer that has the property set to that value.

Ghostkeeper, 2016-07-13:
I would've liked to see something like an extra plane through the build volume at the gantry height while the original build volume is still shown, with a cross through the plane in the same lines as the edges of the build volume, or something like that. But this solution is fine as far as I'm concerned.