Ultimaker/Cura #872: Materials settings are not taken into account

murphy2712, 2016-07-12:
I want to use the material profiles to set the hotend and bed temperatures depending on the filament material I use.

I tried changing the values in Cura.app/.../materials/pla.cfg but when I select the material in Cura, the values don't change according to the file values.

```
[general]
version = 1
type = material
name = PLA

[settings]
material_print_temperature = 211
material_bed_temperature = 71
```

[image]

I also tried creating a new file for a new material (in Cura.app and tried in ~/.cura/profiles too) and while the new material shows in the Cura material list, selecting it does not update any values.

I'm using Cura 2.1.2 on OSX, with a custom Reprap profile based on fdmprinter.json.

fieldOfView, 2016-07-12:
What printer do you use? For the UM2+ the materials in resources/materials are not used. Instead you have to edit the values in resources/profiles/ultimaker 2+

murphy2712, 2016-07-12:
I'm using a custom Reprap printer (SmartrapCore).
Any way to get those working on a non-UM machine?

fieldOfView, 2016-07-12:
Hmm, for a normal reprap machine those materials should be at least somewhat functional. 

Can you use profiles instead of materials for now? There has been an extensive overhaul of this in 2.2, but it will take us a bit longer to psuh out a 2.2 release.

murphy2712, 2016-07-12:
Yes that's what I'm doing in the meantime...

Ghostkeeper, 2016-07-13:
Since this is fixed in 2.2 (as far as we know now) and we have a workaround for 2.1, I'll consider this one fixed :wink:

closed 2016-07-13 by Ghostkeeper

murphy2712, 2016-07-14:
Is there a 2.2 beta build out there? Or should I build it myself to have one?

nallath, 2016-07-15:
There is no beta build out. We do have a system with nightlies, but those are not automatically published.

Ghostkeeper, 2016-07-15:
@thopiekar has a public build of 2.2, but as far as I understood his is a bit old since his build environment broke a bit.

thopiekar, 2016-07-15:
@Ghostkeeper: Got here 2.2 build for Ubuntu here, but @murphy2712 is using OSX or is there something I overlooked?

thopiekar, 2016-07-15:
@murphy2712 If you are using Ubuntu then it is very simple to install my software packages with 2.2 aka built from [master].

Ghostkeeper, 2016-07-15:
Ah right, just PPAs of course. Sorry.

thopiekar, 2016-07-15:
@Ghostkeeper No problem :wink: