Ultimaker/Cura #873: Rotate by green ring doesn't output correct code

NigelWebbZUK, 2016-07-13:
Cura 2.1.2
When I open a STL file, click on it, select rotate, select green ring and drag to rotate by 180 (do not touch any other ring even slightly). The model rotates on the screen but if you produce gcode it is not rotated.

A similar problem can also be observed if after rotating by 180 deg by the green ring, if you select the object and click duplicate, the new object is 'as loaded', ie without rotation.

Nigel

Ghostkeeper, 2016-07-15:
I couldn't reproduce this. I tried on 2.1.3 as well as on the current master version. The g-code is produced of the rotated model.

The problem with the duplicating is known to us though. It has been fixed for version 2.2.

3DAmsterdam, 2016-10-18:
I have seen this too, After reloading the objects it seem to work fine.

nallath, 2017-07-27:
I can't reproduce this. Since then we also redid parts of code that handled this, so it's most likely fixed.

closed 2017-07-27 by nallath