Ultimaker/Cura #876: Very slow slicing (13% CPU usage)

cekuhnen, 2016-07-16:
Hi gents,

I noticed that with more complex designs the slicing and then specifically the layer preview can take a lot of time to calculate.

Here is an example - it also nearly fills the Ultimaker build envelope so it is normal that it takes longer.
[image]

But I noticed that the software seems not to make use of multiple CPUs and cores.
[image]

Can the software be enhanced to make use of those? It would make the slicing calculations so much faster.

While writing this text incl. making screenshoots the layer preview just reached 57%.
[image]

Ghostkeeper, 2016-07-16:
There's an improvement to this in the works as we speak: https://github.com/Ultimaker/Cura/pull/800

JustAnother1, 2016-07-17:
Could it be that your PC has 8 cores?

If so then Cura already uses more than 100% of one core. I assume that curaEngine (doing the actual slicing) is taking one core at 100% the rest is Cura(the GUI) just sitting there.

Slicing is a process of several steps that depend on each other. So splitting these up onto several tasks to have more than one core working on it is not intuitive. The results then also need to be combined again.

I wouldn't expect an multi thread slicer any time soon.

daid, 2016-07-17:
> I wouldn't expect an multi thread slicer any time soon.

Funny, we had a meeting about this last week. There are some parts that can be optimized with multi-core. But without many changes, only about 30% of the process can currently be enhanced.

cekuhnen, 2016-07-17:
Yes it is a Xeon.

I could swear however that (I am printing a former part now) that the slicer got slower.

It takes a lot longer not for the layers to show.

daid, 2016-07-17:
With Cura 2, a lot of features where added, some of these make the process a bit slower.

But what you are most likely seeing, is that the layer view isn't calculating anything to show till the slicing is done. While with the old Cura, it started to show things soon as the data was available.

JustAnother1, 2016-07-17:
thanks @daid.
Is there a chance to shed some more light on those "30%" ?  

I don't want you to spill any company secrets, but I think facts like this should be known to many. That helps the community find a solution or a work around.

Ghostkeeper, 2016-07-17:
Nothing secret about that. Let's hear your ideas.

There's two pieces of the engine that can fairly easily be parallelised: The slicing, which breaks the 3D model up into 2D polygons, and the insets. These together account for about 30-40% of the computation time (depending on whether you want supports or not). Both of these are simply computed layer-by-layer without any interdependencies, and as such they can simply be parallel-looped across different threads.

After that there's the support generation, in which layers are generated top-to-bottom and each layer is dependent on the previous one, so this is difficult to parallelise. I'd love to hear any ideas on this. Then there is the g-code planning and the g-code writing (the bulk of the work). This consists of lots of individual pieces, most of which are dependent on at least some previous parts. Currently these "pieces" are fairly intermixed though, so pulling them apart is quite a risk.

We had other ideas as well but those are all more difficult to implement:
* Make the planning and writing phase start working as soon as we're done processing the lower layers, with a producer-consumer model. This doesn't work if there is support since support is generated top-down.
* The g-code planning and the g-code writing could work with a producer-consumer type structure too. We're not entirely sure what the interdependencies of these two are.
* The g-code planning and writing phase could be made into a pipes-and-filter type structure with tasks that depend on each other. This is extremely challenging to implement though, for the aforementioned reason that they are so intermixed.
* The layer view data could already be sent earlier. This means that the layer view gets less information though. For instance, you could send the layer view data after infill paths are generated (no travel paths), after support is generated (no infill lines), or even after the insets (no support either). We could probably fairly easily do this but it requires some thought on what to display in the GUI then.

Did I summarise our meeting well?

awhiemstra, 2016-07-18:
In regards to:

> The layer view data could already be sent earlier. This means that the layer view gets less information though. For instance, you could send the layer view data after infill paths are generated (no [...]

There are a bunch of constraints the layer view needs to account for. However, most (all?) of these can be taken care of on the frontend side. Additionally, since the frontend already considers the process asynchronous, there is nothing blocking us from sending the data as soon as possible.

fieldOfView, 2016-07-18:
> there is nothing blocking us from sending the data as soon as possible.

(...apart from the lack of time to implement it)

JustAnother1, 2016-07-20:
The dependence on previous calculations is the reason I speculated that a parallel Slicer is still far away.
A chance I see is that re organising the steps based on these interdependencies could work.

Example: The G-Code generation depends on the previous layer to know the start point for the next layer. This could be solved if it is possible to modify the layer processing so that the first step is to calculate the position where this Layer will end. Starting from this point of time another Task running on another core can start processing the next layer. 

So if the process done per layer consist of the steps A, B, C and D and if after A the nozzle position at the end of that layer is known. Then it could be done like this:

Time   |	0  |1     |2	|3	|4|	5|	6|
---------|------|------|------|------|------|------|------|
Task1|	A|	B|	C|	D|	X|	X|	X|
Task2|	|	A|	B|	C|	D|	X|	X|
Task3|	|	|	A|	B|	C|	D|	X|
Task4|	|	|	|	A|	B|	C|	D|

(IN the X slots a new Layer can be processed.I hope this is clear)

Another was of rearranging the work would be to put the needed steps into 2 groups. One group just depends on the Information the Layer has. The other depends on Information about prevous or following layers.

With such a split all cores could be used to execute the group 1 actions for all the layers. Only the group2 actions would then needed to be executed sequential.

But I would rather discuss this based on the details of the implementation. Last time I checked long time ago (Sorry have been busy) there was very little documentation. Is there now any documentation that list these steps and their dependencies?  This might be a great place : https://github.com/Ultimaker/CuraEngine/wiki Or does everybody need to study the code to extract all this valuable information? I don't know when I will find the time to do that (again sorry).

Ghostkeeper, 2016-07-21:
One of the results of that meeting Daid spoke of was that @BagelOrb would make an overview of the dependencies of different parts of the program ;) He hasn't made one yet, as far as I'm aware.

BagelOrb, 2016-07-21:
I will produce such a diagram in the very near future. Any suggestions on what program I should use to generate such a diagram?

nickthetait, 2016-07-21:
Could be valuable to include this diagram with source code. This will allow periodic updates (and historic references) as the architecture evolves. So I would suggest open source tools like Inkscape or LibreOffice Draw.

JustAnother1, 2016-07-21:
If you want to go with the "integrate with source code" route then doxygen makes sense. With the right tools (dot graphviz) it can make nice diagrams of dependencies.
But be aware: If you go that route you are on the level of variable names and file names. If you want to explain what you variable bob is used for then that is the right tool. If you want to give people that don't know the code an overview of the structure behind the code then it is the wrong tool!

For em this seems to be more a "general" discussion. If we come up with solutions then probably a lot of code needs to change. That is for me a reason to not go the "integrate with source code" way.

For overviews use the tool that you can work with best. As long as you can put in Text and pictures, you are good. If you want to get feedback then develop it in the open. If you want others to help you with the work then also use open source tools and have a way to accept changes. Here a Wiki comes to mind. This one is pretty good: https://www.dokuwiki.org/dokuwiki#

Most important: As soon as you start working on this: Let us know! And let us see! (Add a disclaimer that it is not finished, if you must. Ignore comment as long as you need. But don't work in private for weeks and then come up with something odd)

BagelOrb, 2016-07-22:
Thanks a lot for the feedback!

I think I will use dot graphviz, but not via doxygen.

There already is an old (outdated?) diagram to show the setting inheritance
the engine employs. I think it's included in the doxygen somewhere.

I will put the diagram in the source somewhere.

mokusolo, 2016-09-01:
A problem about slicing have troubled me these days.
I loaded a large stl file to Cura 2.1.2. It performed well and in 4 minutes it finished slicing.
But when I installed the latest(nearly,few days ago) CuraEngine and tried to slice it with command line(-j "fdmprinter.def.json").To tell the truth,I didn't finished that for several times.It takes too long.
I know some settings like layer height will affect that.Except that,I just can't figure out why.I'm confused.
Appreciate that if someone could give me some tips.Thanks a lot.

BagelOrb, 2016-09-01:
Did you build the newest in debugging mode perhaps?

Could you provide a link to the STL please?

mokusolo, 2016-09-02:
Yes.A small stl is ok.
The file is here. https://drive.google.com/open?id=0B-2EJkiqetWfNzdQM3Z2TjhYN2c
Thank you for your reply.

jackha, 2016-09-13:
We tested it with the voronoi lamp stl and we cannot reproduce the "slow new version". It even seems slightly faster.
CuraEngine compiled in debug mode can be very slow, this is one of the things first to check. Are there other variables to check?
I strongly suspect the 13.8% is indeed about 1/8 of a quad core + hyperthreading machine.

Our findings:

Testing on Windows 64 using the supplied model (test_57.stl):
Slicing on 2.1: 9m0.408s 8m4.887s
Slicing on 2.3: 8m3.761s 7m6.425s
Slicing on master: 8m4.874s 7m6.517s

Me:
win64, 60% size, normal settings. Single try, so allow for some noise.
2.1: 2:40
2.3 beta: 1:55
master: 2:02

closed 2016-09-13 by nallath