Ultimaker/Cura #883: Startup Changelog dialog issues

juj, 2016-07-23:
After installing Cura 2.1.2 64 bit and starting up, there's some (minor) issues that I notice with the Changelog dialog that greets the user:

[image]

a) The Close button overlaps with the horizontal scrollbar. This might have something to do with high DPI displays, I'm running on a 4K monitor with Windows display scaling set to 150%

b) The dialog is very small, tiny really, but there is a lot of text in it. Why not make the dialog, say, 4x bigger?

c) For some reason, scrolling the dialog is very slow and sluggish. After scrolling the mouse wheel, it takes about a second for the dialog to actually scroll.

fieldOfView, 2016-07-25:
AFAIK, we're working on fixing this in Cura 2.2. 
Are you running Cura on a desktop or a laptop? Is the 4K monitor your only connected display?

juj, 2016-07-27:
The part c) ended up being #884, and no longer occurring after rebooting. I'm running on a Windows 10 with Intel 5960X and 2xSLI GTX980Ti. The 4K monitor is the only connected display. Also, I now notice that b) was not so much of an issue, since the dialog is after all resizable, which I didn't notice at first due to the lagginess trouble I had with c).

nallath, 2017-07-27:
This should already be fixed.

closed 2017-07-27 by nallath