Ultimaker/Cura #884: Cura 2.1.2 64-bit UI is very laggy

juj, 2016-07-23:
Trying out Cura 2.1.2 64-bit on Windows 10 on an 8-core Intel Core i7-5960X CPU and 32GB of RAM, the UI of the software acts sluggishly/with a lot of lag. Holding down right mouse button, or choosing any menu items has a latency of 1-2 seconds. This occurs already before any models have been loaded into the application.

Observing the application in Windows Task Manager shows that the application constantly takes up roughly one full application core of CPU time on this system (100%/16 logical cores = 6.25%):

[image]

juj, 2016-07-23:
Looking in Task Manager and waiting for Cura.exe to "settle", it seems that the process occasionally spins down to 0% of usage, but every few seconds or so spins back up to holding one full core (6-7% of CPU usage).

awhiemstra, 2016-07-25:
Looking also at the responsiveness of #883 it seems to suggest something is wrong with your OpenGL. What video card are you using? First thing to try is to upgrade its drivers to the most recent version.

juj, 2016-07-27:
This was on a 2x GTX 980 Ti in SLI configuration with the latest NVidia drivers (works ok in Doom in OpenGL 4.5 mode). Restarting the application did not help fix performance, but I now noticed that rebooting the computer after installation did. I seldomly reboot the computer so it's on for several days at a time, sleeping in between. Before rebooting, other applications did work with good performance. Not sure what was causing it, but closing since this ended up being transient. Thanks for the quick reply!

closed 2016-07-27 by juj