Ultimaker/Cura #886: cura mirrors everything

scarabr, 2016-07-25:
my printer mirrors everything on my printer prusa i3 and its caught me off guard on 3 prints is there any setting to change or adjustment? either my printer or cura

thopiekar, 2016-07-25:
Could you please make screenshots and maybe photos of your prints?

Would be useful to understand what gets "mirrored". A different user had almost the same problem, but she/he noticed that he has his stepper motor for Y connected wrong.
So please double check that your printer is working correct - just to be sure.

Ghostkeeper, 2016-07-25:
If the g-code is showing mirrored, then it's Cura's fault. You can check that in Layer View in Cura, or with an actual g-code viewer such as RepetierHost. If the g-code is not mirrored, something is up either with the way your motor is connected or with the firmware.

Cura doesn't mirror objects through settings. It can only mirror it with the mirror tool, but then it shows up mirrored in the 3D view as well so you know what you're printing.

thopiekar, 2016-09-17:
@scarabr, Please reopen if still happening..

closed 2016-09-17 by thopiekar