Ultimaker/Cura #887: [master] qmlRegisterSingletonType(): "MachineManager" is not available because the callback function returns a null pointer.

thopiekar, 2016-07-25:
Currently appears when using [master] branches:

```
[35m2016-07-25 13:14:25,073[0m - [36mDEBUG[0m - [37m.../UM/Controller.py (setActiveView [80]): Setting active view to SolidView[0m
[35m2016-07-25 13:14:25,720[0m - [31m[47mCRITICAL[0m - [37m.../cura/CrashHandler.py (show [24]): An uncaught exception has occurred![0m
[35m2016-07-25 13:14:25,722[0m - [31m[47mCRITICAL[0m - [37m.../cura/CrashHandler.py (show [24]): Traceback (most recent call last):[0m
[35m2016-07-25 13:14:25,722[0m - [31m[47mCRITICAL[0m - [37m.../cura/CrashHandler.py (show [24]):   File "/home/thopiekar/Projekte/eclipse-Ultimaker/Cura/cura/CuraApplication.py", line 436, in getMachineManager[0m
[35m2016-07-25 13:14:25,723[0m - [31m[47mCRITICAL[0m - [37m.../cura/CrashHandler.py (show [24]):     self._machine_manager = cura.Settings.MachineManager.createMachineManager()[0m
[35m2016-07-25 13:14:25,723[0m - [31m[47mCRITICAL[0m - [37m.../cura/CrashHandler.py (show [24]):   File "/home/thopiekar/Projekte/eclipse-Ultimaker/Cura/cura/Settings/MachineManager.py", line 809, in createMachineManager[0m
[35m2016-07-25 13:14:25,723[0m - [31m[47mCRITICAL[0m - [37m.../cura/CrashHandler.py (show [24]):     return MachineManager()[0m
[35m2016-07-25 13:14:25,723[0m - [31m[47mCRITICAL[0m - [37m.../cura/CrashHandler.py (show [24]):   File "/home/thopiekar/Projekte/eclipse-Ultimaker/Cura/cura/Settings/MachineManager.py", line 62, in __init__[0m
[35m2016-07-25 13:14:25,723[0m - [31m[47mCRITICAL[0m - [37m.../cura/CrashHandler.py (show [24]):     self.setActiveMachine(active_machine_id)[0m
[35m2016-07-25 13:14:25,724[0m - [31m[47mCRITICAL[0m - [37m.../cura/CrashHandler.py (show [24]):   File "/home/thopiekar/Projekte/eclipse-Ultimaker/Cura/cura/Settings/MachineManager.py", line 361, in setActiveMachine[0m
[35m2016-07-25 13:14:25,724[0m - [31m[47mCRITICAL[0m - [37m.../cura/CrashHandler.py (show [24]):     Application.getInstance().setGlobalContainerStack(containers[0])[0m
[35m2016-07-25 13:14:25,724[0m - [31m[47mCRITICAL[0m - [37m.../cura/CrashHandler.py (show [24]):   File "/home/thopiekar/Projekte/eclipse-Ultimaker/Uranium/UM/Application.py", line 123, in setGlobalContainerStack[0m
[35m2016-07-25 13:14:25,724[0m - [31m[47mCRITICAL[0m - [37m.../cura/CrashHandler.py (show [24]):     self.globalContainerStackChanged.emit()[0m
[35m2016-07-25 13:14:25,724[0m - [31m[47mCRITICAL[0m - [37m.../cura/CrashHandler.py (show [24]):   File "/home/thopiekar/Projekte/eclipse-Ultimaker/Uranium/UM/Signal.py", line 141, in emit[0m
[35m2016-07-25 13:14:25,725[0m - [31m[47mCRITICAL[0m - [37m.../cura/CrashHandler.py (show [24]):     func(dest, *args, **kwargs)[0m
[35m2016-07-25 13:14:25,725[0m - [31m[47mCRITICAL[0m - [37m.../cura/CrashHandler.py (show [24]):   File "/home/thopiekar/Projekte/eclipse-Ultimaker/Cura/cura/BuildVolume.py", line 256, in _onGlobalContainerStackChanged[0m
[35m2016-07-25 13:14:25,725[0m - [31m[47mCRITICAL[0m - [37m.../cura/CrashHandler.py (show [24]):     self._updateDisallowedAreas()[0m
[35m2016-07-25 13:14:25,725[0m - [31m[47mCRITICAL[0m - [37m.../cura/CrashHandler.py (show [24]):   File "/home/thopiekar/Projekte/eclipse-Ultimaker/Cura/cura/BuildVolume.py", line 314, in _updateDisallowedAreas[0m
[35m2016-07-25 13:14:25,725[0m - [31m[47mCRITICAL[0m - [37m.../cura/CrashHandler.py (show [24]):     bed_adhesion_size = self._getBedAdhesionSize(self._active_container_stack)[0m
[35m2016-07-25 13:14:25,726[0m - [31m[47mCRITICAL[0m - [37m.../cura/CrashHandler.py (show [24]):   File "/home/thopiekar/Projekte/eclipse-Ultimaker/Cura/cura/BuildVolume.py", line 369, in _getBedAdhesionSize[0m
[35m2016-07-25 13:14:25,726[0m - [31m[47mCRITICAL[0m - [37m.../cura/CrashHandler.py (show [24]):     skirt_size = container_stack.getProperty("brim_line_count", "value") * container_stack.getProperty("skirt_brim_line_width", "value")[0m
[35m2016-07-25 13:14:25,726[0m - [31m[47mCRITICAL[0m - [37m.../cura/CrashHandler.py (show [24]): TypeError: unsupported operand type(s) for *: 'int' and 'NoneType'[0m
qmlRegisterSingletonType(): "MachineManager" is not available because the callback function returns a null pointer.
```

fieldOfView, 2016-07-25:
Looks like it is related to this commit: https://github.com/Ultimaker/Cura/commit/d4f96ab62daa4e573c3ec53e224cfed8d3f912dd
Could you try again with a clean .local/shared/cura profile?

thopiekar, 2016-07-25:
Of course I can :+1:

thopiekar, 2016-07-25:
Oh, well.. Now my PPA code returns:
```
2016-07-25 14:30:46,716 - DEBUG - .../UM/Controller.py (setActiveView [80]): Setting active view to SolidView
2016-07-25 14:30:47,893 - DEBUG - .../UM/Mesh/MeshData.py (calculateNormalsFromVertices [339]): Calculating normals took 0.002471446990966797 seconds
2016-07-25 14:30:47,894 - DEBUG - /usr/lib/uranium/plugins/FileHandlers/STLReader/STLReader.py (read [35]): Loaded a mesh with 22626 vertices
2016-07-25 14:30:47,902 - CRITICAL - .../cura/CrashHandler.py (show [24]): An uncaught exception has occurred!
2016-07-25 14:30:47,907 - CRITICAL - .../cura/CrashHandler.py (show [24]): Traceback (most recent call last):
2016-07-25 14:30:47,907 - CRITICAL - .../cura/CrashHandler.py (show [24]):   File "/usr/lib/python3/dist-packages/cura/CuraApplication.py", line 424, in getMachineManager
2016-07-25 14:30:47,908 - CRITICAL - .../cura/CrashHandler.py (show [24]):     self._machine_manager = cura.Settings.MachineManager.createMachineManager()
2016-07-25 14:30:47,908 - CRITICAL - .../cura/CrashHandler.py (show [24]):   File "/usr/lib/python3/dist-packages/cura/Settings/MachineManager.py", line 702, in createMachineManager
2016-07-25 14:30:47,908 - CRITICAL - .../cura/CrashHandler.py (show [24]):     return MachineManager()
2016-07-25 14:30:47,908 - CRITICAL - .../cura/CrashHandler.py (show [24]):   File "/usr/lib/python3/dist-packages/cura/Settings/MachineManager.py", line 60, in __init__
2016-07-25 14:30:47,908 - CRITICAL - .../cura/CrashHandler.py (show [24]):     self.setActiveMachine(active_machine_id)
2016-07-25 14:30:47,908 - CRITICAL - .../cura/CrashHandler.py (show [24]):   File "/usr/lib/python3/dist-packages/cura/Settings/MachineManager.py", line 268, in setActiveMachine
2016-07-25 14:30:47,909 - CRITICAL - .../cura/CrashHandler.py (show [24]):     Application.getInstance().setGlobalContainerStack(containers[0])
2016-07-25 14:30:47,909 - CRITICAL - .../cura/CrashHandler.py (show [24]):   File "/usr/lib/python3/dist-packages/UM/Application.py", line 122, in setGlobalContainerStack
2016-07-25 14:30:47,909 - CRITICAL - .../cura/CrashHandler.py (show [24]):     self.globalContainerStackChanged.emit()
2016-07-25 14:30:47,909 - CRITICAL - .../cura/CrashHandler.py (show [24]):   File "/usr/lib/python3/dist-packages/UM/Signal.py", line 141, in emit
2016-07-25 14:30:47,909 - CRITICAL - .../cura/CrashHandler.py (show [24]):     func(dest, *args, **kwargs)
2016-07-25 14:30:47,909 - CRITICAL - .../cura/CrashHandler.py (show [24]):   File "/usr/lib/python3/dist-packages/cura/BuildVolume.py", line 192, in _onGlobalContainerStackChanged
2016-07-25 14:30:47,910 - CRITICAL - .../cura/CrashHandler.py (show [24]):     self._updateDisallowedAreas()
2016-07-25 14:30:47,910 - CRITICAL - .../cura/CrashHandler.py (show [24]):   File "/usr/lib/python3/dist-packages/cura/BuildVolume.py", line 218, in _updateDisallowedAreas
2016-07-25 14:30:47,910 - CRITICAL - .../cura/CrashHandler.py (show [24]):     skirt_size = self._getSkirtSize(self._active_container_stack)
2016-07-25 14:30:47,910 - CRITICAL - .../cura/CrashHandler.py (show [24]):   File "/usr/lib/python3/dist-packages/cura/BuildVolume.py", line 282, in _getSkirtSize
2016-07-25 14:30:47,910 - CRITICAL - .../cura/CrashHandler.py (show [24]):     skirt_size = container_stack.getProperty("brim_line_count", "value") * container_stack.getProperty("skirt_line_width", "value")
2016-07-25 14:30:47,910 - CRITICAL - .../cura/CrashHandler.py (show [24]): TypeError: unsupported operand type(s) for *: 'int' and 'NoneType'
qmlRegisterSingletonType(): "MachineManager" is not available because the callback function returns a null pointer.
Abgebrochen (Speicherabzug geschrieben)
```

But [master] works :sweat_smile: 
Is that behavior "correct"?

fieldOfView, 2016-07-25:
I think that your PPA and current master cannot use eachothers configurations. We will support going from 2.1 to 2.2, but while master is in flux compatibility of configuration files is not guaranteed; you may have to clear your profile every now and then. Such is the disadvantage of the bleeding edge.

thopiekar, 2016-07-25:
Ok, thanks!

closed 2016-07-25 by thopiekar