Ultimaker/Cura #888: [master] Cura tries to import logs as preferences

thopiekar, 2016-07-25:
I guess the log says enough:

```
[35m2016-07-25 13:14:22,247[0m - [33mWARNING[0m - [37m.../UM/VersionUpgradeManager.py (_upgradeFile [207]): Invalid preferences file: /home/thopiekar/.local/share/cura/././cura.log[0m
```

fieldOfView, 2016-07-25:
Same for plugins, if the user stores plugins in their profile folder (instead of in the application resources)

thopiekar, 2016-07-25:
Now after removing .local/share/cura it still tries to load settings from "/home/thopiekar/.local/share/cura/././cura.log" == "/home/thopiekar/.local/share/cura/cura.log".
Can't believe this is correct :eyes:

Ghostkeeper, 2016-07-25:
This was introduced when the version upgrade system needed to be able to handle dynamic types of preferences due to the setting rework. There's three types of preferences that it processes.

* Profiles to quality profiles. Profiles are stored in the <share>/cura/profiles folder, or any subfolder.
* Machine instances to container stacks. Machine instances are stored in the <share>/cura/machine_instances folder.
* Preferences to preferences. Preferences are stored in the <share>/cura folder.

Since the version upgrade system doesn't know about these preference types, it simply looks in the folder that the plug-in specifies and all subfolders for files that it can convert and tries to convert them all.

The only fix that I see is to add more specific data to the plug-in on where to find the files it converts.

awhiemstra, 2016-07-26:
> The only fix that I see is to add more specific data to the plug-in on where to find the files it converts.

Or, you know, just silently ignore them instead of spitting out a warning.

thopiekar, 2016-07-26:
Or creating a blacklist of files or file extensions, which would never
include any preferences 🤔

BagelOrb, 2017-07-21:
no info provided for too long

closed 2017-07-21 by BagelOrb