Ultimaker/Cura #892: Embed Machineface into Cura - need some help

strahlex, 2016-07-26:
I would like to combine Machineface and Cura. [Machineface](https://github.com/strahlex/Machineface) is a 3D printing remote UI for Machinekit, the generic open source machine control software, based on [QtQuickVcp](https://github.com/strahlex/QtQuickVcp).

So far I created a custom config for my 3D printer and extended the PostProcessing plugin to convert RepRap GCode to Machinekit compatible GCode. Now I would to embed Machineface in Cura itself, should be an easy task since it is based on Qt5 too.

I am very familiar with working with Qt. However, the Cura build system seems to be based cmake. There are no hints how to build Cura in the `README.md`. Can you please point me where to find some instructions to get started.

Moreover, how could I deploy/publish my work as plugin?

Would be great if you could help me.

thopiekar, 2016-07-26:
If you are on Ubuntu you can try my PPA based on [master] branches:
https://launchpad.net/~thopiekar/+archive/ubuntu/cura-master

You'll also find an updated PyQt5 package there. If you also need a newer Qt5 package I'll test whether it makes sense (while thinking about all the other Qt-based software) to upgrade Qt5 here, too.
Building a .deb package via "cura-build" is the official way, but, like mentioned above, you can use my PPA, too. These packages are updated at least every week.
If you want to code in Cura directly, you can clone Cura and Uranium (optionally) and run cura_app.py in your Cura clone. If you've made modifications to Uranium, don't forget to set "PYTHONPATH=<path_to_my_own_Uranium_copy>" to use your own copy.
You can take this as an example to install your plugin later: https://github.com/nallath/PostProcessingPlugin/blob/master/CMakeLists.txt

Finally and most important, take a look at "MachineAction".
https://github.com/Ultimaker/Cura/search?utf8=%E2%9C%93&q=MachineAction&type=Code
I'm sure someone else can give you more detailed info on them if needed.

Ghostkeeper, 2016-07-27:
It would be ideal if you were to write a Cura plug-in out of this. To turn a folder into a Cura plug-in, it needs to contain an `__init__.py` file with a `getMetaData()` function. You can take an example from [another plug-in](https://github.com/Ultimaker/Cura/blob/master/plugins/UltimakerMachineActions/__init__.py) for specifics or refer to our guide (the examples are slightly outdated): https://github.com/Ultimaker/Uranium/blob/master/docs/plugins.md
I suggest you define a `machine_action` type plug-in, like the above example.

The work would be in adding actions to these buttons. They currently seem to be no-op (as it is user interface only). You'd have to add a Python file that acts as a model for the QML. It would need to be registered (like [this](https://github.com/Ultimaker/Cura/blob/master/cura/CuraApplication.py#L415)), and it would need a function with an `@pyqtSlot` annotation to allow QML to communicate g-code commands to Python. That annotated function can then call upon the [USBPrinting](https://github.com/Ultimaker/Cura/tree/master/plugins/USBPrinting) plug-in or re-implement the serial connection code if calling the plug-in proves a problem due to the nature of our plug-ins (as in, you are not guaranteed that the plug-in exists or is active and our plug-in system has no way to define dependencies).

The video tab could also become a hurdle. I'm sure @nallath could help out if he finds the time, since he implemented video communication multiple times for unrelated projects.

Ghostkeeper, 2016-07-27:
As for building Cura, we are just in the transition from one system to another, though both are using CMake. I don't know if we're currently still using the [cura-build](https://github.com/Ultimaker/cura-build) repository or that we've completed the [cura-build-enviroment](https://github.com/Ultimaker/cura-build-environment) repository yet, but I think it's still just cura-build.

awhiemstra, 2016-07-27:
For the moment, you should still use cura-build. cura-build-environment does not contain actual Cura, just its dependencies.

strahlex, 2016-07-28:
Thanks, got it partially working.

How can I add new file type/extensions to Cura so they can be used in the 3d-printer JSON file? In particular I need .ngc (RS274 Gcode) which is the file type used by Machinekit/LinuxCNC

Ghostkeeper, 2016-07-29:
We use file writer plug-ins such as this one: https://github.com/Ultimaker/Cura/tree/master/plugins/GCodeWriter

strahlex, 2016-07-29:
@Ghostkeeper Thanks, thats exactly what I was looking for.

machinekoder, 2016-09-11:
I have created a [blog post](http://machinekoder.com/?p=9) about this topic.

nallath, 2017-07-27:
Closing due to inactivity

closed 2017-07-27 by nallath