Ultimaker/Cura #894: machine_center_is_zero Causes Freeze

dr2391, 2016-07-27:
Just created a JSON file for my new Delta printer but every time I tried to slice a couple different objects that usually take 2-4 hours to print, Cura freezes up about half way through and I have to force quit. Smaller 4 min prints sliced fine.

Through process of elimination, I found that by removing or setting machine_center_is_zero to false, Cura slices the files with no problem.

I'm using the newest version of Cura

Can anyone reproduce this?

Ghostkeeper, 2016-07-27:
I mistakenly tried to reproduce this issue on Cura 2 master (Ubuntu) when you mistakenly posted this issue in daid's repository. And I couldn't reproduce it. I also tried on Cura 2.1.1 (Windows x64) just now, and no luck either. Just slices, and I'm using quite large models (takes more than a day to print).

Perhaps without publicising details on your new printer, could you give an example of a model that fails to slice?

Of course, perhaps the reason why I can't reproduce it is because I couldn't test on OSX.

dr2391, 2016-07-27:
Thanks for helping Ghostkeeper. Sorry about the the post in daid's repo. 

Here's a file that will slice when `machine_center_is_zero` is `false`. When set to `true`, Cura locks up about halfway through slicing.

[mouthpiece.stl.zip](https://github.com/Ultimaker/Cura/files/387394/mouthpiece.stl.zip)

dr2391, 2016-07-28:
Here's the log during slicing:
```
DEBUG - Autosaving preferences, instances and profiles
DEBUG - Calculating normals took 0.002819061279296875 seconds
DEBUG - Loaded a mesh with 34158 vertices
DEBUG - Loading mesh took 1.3033950328826904 seconds
DEBUG - Attempting to kill the engine process
DEBUG - Killing engine process
DEBUG - Engine process is killed. Recieved return code -15
INFO - Backend crashed or closed. Restarting...
DEBUG - Attempting to kill the engine process
INFO - Started engine process: /Applications/Cura.app/Contents/MacOS/CuraEngine
DEBUG - Backend connected on port 49674
DEBUG - Backend quit with return code -11. Resetting process and socket.
```

Ghostkeeper, 2016-07-28:
Still slices fine for me on Cura 2.1 on Ubuntu, using your model in the Malyan M180 printer (which has `machine_center_is_zero` set to `true`). Must be OSX only :(

dr2391, 2016-07-30:
Just upgraded to OS X 10.11.6 and still having the same problem

ytotsuka2305, 2016-09-01:
I is also the same problem.
I use Custome FDM Printer.
Set in the true the machine_center_is_zero, cura_engin crashes.

Process:               CuraEngine [85749]
Path:                  /Users/USER/*/Cura.app/Contents/MacOS/CuraEngine
OS Version:            Mac OS X 10.10.5 (14F1909)
Exception Type:        EXC_CRASH (SIGABRT)
Exception Codes:       0x0000000000000000, 0x0000000000000000

Application Specific Information:
Assertion failed: (false && "We should have returned already before here!"), function insert, file /Volumes/*/Users/*/cura-build-master/build/CuraEngine-prefix/src/CuraEngine/src/utils/SparseLineGrid.h, line 112.

markwal, 2016-09-01:
I also hit this assert. I just reverted the commit that added it and I work fine. Been meaning to look closer at it to propose a pr since @bagelorb suggested, but haven't got round to it yet.

markwal, 2016-09-01:
BTW doesn't hit on every model, just most of the time with 0, 0 at the center.

ytotsuka2305, 2016-09-02:
Thanks for markwal.
It will be successful and to place the object in the Y-axis -20 or more.
X0,Y0 Crash.
X20,Y-20 Ok.
X30,Y20 Crash.
X-50,Y50 Crash.
X20,Y-40 Ok.

ytotsuka2305, 2016-09-06:
added screenshot.
[image]
[image]

BagelOrb, 2016-09-06:
Mark fixed this in the meantime.

Marks fix will be in the next release.

closed 2016-09-06 by BagelOrb