Ultimaker/Cura #897: wont slice

cartmailj, 2016-07-29:
i have added another item to the print bed and it simply wont slice the print

fieldOfView, 2016-07-29:
Could you give us a _bit_ more info? "it simply wont slice" gives us very little to go on.

What version of Cura are you using, on what computer, and with what printer?
How big are your objects? Are both of them shown in yellow or are they grey?
Have you made any changes to the settings? Are you by any chance using "one at a time" print sequence?
What does Cura say in the lower right corner? Is it preparing to slice, is it slicing (and failing), or does it say it is done slicing?

cartmailj, 2016-07-29:
sorry i dont know the version of cura its the one shown on your website.
custom built pc fx6300 r9 380 8gb ram  ,the items are both yellow and  it
isnt set to one at a time sequence it says preparing to slice no matter
what i do it stays the same

fieldOfView, 2016-07-29:
Thanks for the additional info. I'm closing this issue because it seems to be a duplicate: 
https://github.com/Ultimaker/Cura/issues/794
Please continue this discussion in that thread.

closed 2016-07-29 by fieldOfView