Ultimaker/Cura #899: Communication problems between Arcus and the engine

thopiekar, 2016-07-30:
Just rebuilt everything in my [master] PPA. So I found this.
Looks like we've missed to commit some changes in the engine or libArcus regarding the protocol.

```
2016-07-30 13:39:43,856 - INFO - .../UM/Backend/Backend.py (startEngine [55]): Started engine process: /usr/bin/CuraEngine
2016-07-30 13:39:43,857 - DEBUG - .../UM/Backend/Backend.py (_backendLog [80]): [Backend] Calling engine with: ['/usr/bin/CuraEngine', 'connect', '127.0.0.1:49676', '-j', '/usr/share/cura/resources/definitions/fdmprinter.def.json', '-vv']
2016-07-30 13:39:43,867 - DEBUG - .../UM/Backend/Backend.py (_backendLog [80]): [Backend] 
2016-07-30 13:39:43,867 - DEBUG - .../UM/Backend/Backend.py (_backendLog [80]): [Backend] Cura_SteamEngine version DEV
2016-07-30 13:39:43,868 - DEBUG - .../UM/Backend/Backend.py (_backendLog [80]): [Backend] Copyright (C) 2014 David Braam
2016-07-30 13:39:43,868 - DEBUG - .../UM/Backend/Backend.py (_backendLog [80]): [Backend] 
2016-07-30 13:39:43,868 - DEBUG - .../UM/Backend/Backend.py (_backendLog [80]): [Backend] This program is free software: you can redistribute it and/or modify
2016-07-30 13:39:43,869 - DEBUG - .../UM/Backend/Backend.py (_backendLog [80]): [Backend] it under the terms of the GNU Affero General Public License as published by
2016-07-30 13:39:43,869 - DEBUG - .../UM/Backend/Backend.py (_backendLog [80]): [Backend] the Free Software Foundation, either version 3 of the License, or
2016-07-30 13:39:43,869 - DEBUG - .../UM/Backend/Backend.py (_backendLog [80]): [Backend] (at your option) any later version.
2016-07-30 13:39:43,869 - DEBUG - .../UM/Backend/Backend.py (_backendLog [80]): [Backend] 
2016-07-30 13:39:43,870 - DEBUG - .../UM/Backend/Backend.py (_backendLog [80]): [Backend] This program is distributed in the hope that it will be useful,
2016-07-30 13:39:43,870 - DEBUG - .../UM/Backend/Backend.py (_backendLog [80]): [Backend] but WITHOUT ANY WARRANTY; without even the implied warranty of
2016-07-30 13:39:43,870 - DEBUG - .../UM/Backend/Backend.py (_backendLog [80]): [Backend] MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
2016-07-30 13:39:43,870 - DEBUG - .../UM/Backend/Backend.py (_backendLog [80]): [Backend] GNU Affero General Public License for more details.
2016-07-30 13:39:43,871 - DEBUG - .../UM/Backend/Backend.py (_backendLog [80]): [Backend] 
2016-07-30 13:39:43,871 - DEBUG - .../UM/Backend/Backend.py (_backendLog [80]): [Backend] You should have received a copy of the GNU Affero General Public License
2016-07-30 13:39:43,871 - DEBUG - .../UM/Backend/Backend.py (_backendLog [80]): [Backend] along with this program.  If not, see <http://www.gnu.org/licenses/>.
2016-07-30 13:39:43,871 - DEBUG - .../UM/Backend/Backend.py (_backendLog [80]): [Backend] Connecting to 127.0.0.1:49676
2016-07-30 13:39:43,929 - DEBUG - .../UM/Backend/Backend.py (_onSocketStateChanged [147]): Backend connected on port 49676
2016-07-30 13:39:43,962 - DEBUG - .../UM/Backend/Backend.py (_backendLog [80]): [Backend] Connected to 127.0.0.1:49676
2016-07-30 13:39:44,712 - DEBUG - .../UM/Backend/Backend.py (_backendLog [80]): [Backend] [DEBUG] Received a Slice message
2016-07-30 13:39:44,713 - DEBUG - .../UM/Backend/Backend.py (_backendLog [80]): [Backend] [ERROR] Warning: setting an unregistered setting material_bed_temp_prepend to True
2016-07-30 13:39:44,713 - DEBUG - .../UM/Backend/Backend.py (_backendLog [80]): [Backend] Loading /usr/share/cura/resources/definitions/fdmextruder.def.json...
2016-07-30 13:39:44,715 - DEBUG - .../UM/Backend/Backend.py (_backendLog [80]): [Backend] [ERROR] Unregistered setting mesh_position_z
2016-07-30 13:39:44,715 - DEBUG - .../UM/Backend/Backend.py (_backendLog [80]): [Backend] [ERROR] Unregistered setting mesh_position_y
2016-07-30 13:39:44,716 - DEBUG - .../UM/Backend/Backend.py (_backendLog [80]): [Backend] [ERROR] Unregistered setting mesh_position_x
2016-07-30 13:39:44,716 - DEBUG - .../UM/Backend/Backend.py (_backendLog [80]): [Backend] [ERROR] Unregistered setting center_object
2016-07-30 13:39:44,716 - DEBUG - .../UM/Backend/Backend.py (_backendLog [80]): [Backend] [DEBUG] Done reading Slice message
2016-07-30 13:39:44,716 - DEBUG - .../UM/Backend/Backend.py (_backendLog [80]): [Backend] [DEBUG] Slicing 1 objects
2016-07-30 13:39:44,717 - DEBUG - .../UM/Backend/Backend.py (_backendLog [80]): [Backend] [DEBUG] Slicing object 1 of 1
2016-07-30 13:39:44,717 - DEBUG - .../UM/Backend/Backend.py (_backendLog [80]): [Backend] [WARNING]  -s wall_thickness="1.2" -s travel="0" -s support_z_distance="0.15" -s skin_overlap="5" -s acceleration_wall="3000" -s acceleration_support="3000" -s wireframe_flow="100" -s acceleration_print="3000" -s machine_settings="0" -s top_thickness="1.2" -s infill_overlap="10" -s meshfix="0" -s raft_acceleration="3000" -s bottom_thickness="1.2" -s retraction_hop_enabled="False" -s cool_fan_speed="100.0" -s machine_head_polygon="[[-1, 1], [-1, -1], [1, -1], [1, 1]]" -s support_extruder_nr="0" -s wall_line_width="0.4" -s line_width="0.4" -s speed="0" -s infill="0" -s raft_fan_speed="0" -s blackmagic="0" -s speed_support="60" -s experimental="0" -s raft_jerk="20" -s top_bottom_thickness="1.2" -s machine_head_with_fans_polygon="[[-20, 10], [10, 10], [10, -10], [-20, -10]]" -s jerk_print="20" -s wireframe_nozzle_clearance="1" -s wireframe_roof_drag_along="0.8" -s wireframe_roof_fall_down="2" -s wireframe_straight_before_down="20" -s wireframe_strategy="compensate" -s wireframe_drag_along="0.6" -s wireframe_top_jump="0.6" -s wireframe_flat_delay="0.1" -s wireframe_flow_flat="100" -s wireframe_flow_connection="100" -s wireframe_printspeed_flat="5" -s wireframe_printspeed_down="5" -s brim_width="8.0" -s wireframe_printspeed_up="5" -s wireframe_printspeed_bottom="5" -s wireframe_height="3" -s magic_fuzzy_skin_thickness="0.3" -s magic_fuzzy_skin_enabled="False" -s support_conical_angle="30" -s support_conical_enabled="False" -s cool_fan_full_at_height="0.2" -s skin_outline_count="0" -s coasting_min_volume="0.8" -s coasting_volume="0.064" -s coasting_enable="False" -s draft_shield_height_limitation="full" -s conical_overhang_enabled="False" -s draft_shield_dist="10" -s draft_shield_enabled="False" -s magic_mesh_surface_mode="normal" -s material="0" -s infill_mesh="False" -s print_sequence="all_at_once" -s meshfix_keep_open_polygons="False" -s meshfix_union_all_remove_holes="False" -s meshfix_union_all="True" -s ooze_shield_enabled="False" -s wireframe_fall_down="0.5" -s retraction_hop_only_when_collides="False" -s magic_fuzzy_skin_point_dist="0.8" -s retraction_count_max="90" -s bottom_layers="6" -s cool_fan_enabled="True" -s retraction_min_travel="0.8" -s retraction_amount="6.5" -s retraction_enable="True" -s material_diameter="1.75" -s machine_max_acceleration_z="100" -s support_join_distance="2.0" -s material_extrusion_cool_down_speed="0.5" -s resolution="0" -s coasting_speed="90" -s infill_pattern="grid" -s acceleration_topbottom="3000" -s material_flow_dependent_temperature="False" -s raft_base_speed="22.5" -s travel_compensate_overlapping_walls_enabled="True" -s infill_before_walls="True" -s switch_extruder_retraction_speed="20" -s infill_sparse_thickness="0.2" -s raft_base_jerk="20" -s jerk_support="20" -s support_infill_rate="15" -s wireframe_top_delay="0" -s speed_support_roof="40.0" -s machine_max_jerk_z="0.4" -s support_type="everywhere" -s wireframe_bottom_delay="0" -s gantry_height="99999999999" -s layer_0_z_overlap="0.15" -s skin_overlap_mm="0.02" -s speed_wall="30.0" -s speed_wall_0="30.0" -s jerk_prime_tower="20" -s speed_print="60" -s skin_no_small_gaps_heuristic="True" -s material_flow="100" -s support_roof_enable="False" -s xy_offset="0" -s jerk_infill="20" -s travel_compensate_overlapping_walls_x_enabled="True" -s machine_disallowed_areas="[]" -s retraction_hop="1" -s machine_use_extruder_offset_to_offset_coords="True" -s wall_0_inset="0" -s speed_wall_x="60.0" -s acceleration_skirt="3000" -s top_bottom_pattern="lines" -s cool_fan_speed_max="100.0" -s retraction_prime_speed="25" -s acceleration_wall_0="3000" -s retraction_speed="25" -s speed_infill="60" -s jerk_support_roof="20" -s wall_line_count="3" -s cool_min_layer_time="5" -s prime_tower_line_width="0.4" -s retraction_extrusion_window="6.5" -s wireframe_enabled="False" -s cool_fan_speed_min="100.0" -s machine_min_cool_heat_time_window="50.0" -s machine_nozzle_tip_outer_diameter="1" -s extruder_prime_pos_z="0" -s raft_base_line_width="0.8" -s machine_nozzle_head_distance="3" -s machine_accelera [... cut ...]
2016-07-30 13:39:45,191 - DEBUG - /usr/lib/cura/plugins/CuraEngineBackend/CuraEngineBackend.py (_terminate [161]): Attempting to kill the engine process
2016-07-30 13:39:45,192 - DEBUG - /usr/lib/cura/plugins/CuraEngineBackend/CuraEngineBackend.py (_terminate [161]): Killing engine process
2016-07-30 13:39:45,193 - DEBUG - /usr/lib/cura/plugins/CuraEngineBackend/CuraEngineBackend.py (_terminate [161]): Engine process is killed. Received return code 0
2016-07-30 13:39:45,194 - ERROR - /usr/lib/cura/plugins/CuraEngineBackend/CuraEngineBackend.py (_onSocketError [240]): A socket error caused the connection to be reset
```

sedwards2009, 2016-07-30:
The socket error at the end suggests that the communication with Arcus was cut off and caused the unknown message type problem. That is a theory, but yes something is not right here.

awhiemstra, 2016-08-01:
It works for me here. Please verify that everything is up to date. I had a similar issue last week that turned out to be an outdated engine.

thopiekar, 2016-08-01:
@awhiemstra: You are right. I'm sure I copied an updated "CuraEngine" executable to my "$WORKBENCH/bin/" but it seems that I missed that.

Created #901 to get a rid of this.

closed 2016-09-17 by thopiekar