Ultimaker/Cura #900: Cura Slice speed vs Craftware slice speed / quality ?

cekuhnen, 2016-08-01:
Craftware works through a design I have in a matter of few seconds
while Cura works over many many many minutes

the time difference is really painfully evident

but I am curious if there is a reason why Cura is so slow? is the gcode better
with a more accurate printing???

Ghostkeeper, 2016-08-01:
We don't rightly know the difference since CraftWare is closed-source.

A lot of the processing time in Cura is devoted to making the print quality better, such as by preventing travel moves over already-printed skin, and avoiding other objects during travel moves. Another source of time constraints is to try and reduce the print time by finding the optimal path, thereby allowing the user to select higher quality settings for the same print time. Typically, slices with Cura spend about 5% of their time in travel moves, on models where Craftware stays around 20% (my test just now, with n=3). Those are some things that I haven't seen CraftWare do very well.

CraftWare does seem to have their multi-threading down very well. Cura is trying to improve this recently, but our current code base is very much focussed on single-threading. This is just speculation, of course. It's hard to make good measurements.

cekuhnen, 2016-08-01:
I would love to be able to do a test print with Craftware but I am not sure if I can just create G code and load it into the Ultimaker2+.

Anybody did that?

Ghostkeeper, 2016-08-01:
I suggest that you copy the header from a Cura-generated g-code file before you give the g-code to an Ultimaker. CraftWare just generates a header it thinks is good for every printer, which messes up the initial unretraction on Ultimaker printers.

cekuhnen, 2016-08-01:
Oh that could be it already? interetsing might have to test that

closed 2016-08-01 by Ghostkeeper