Ultimaker/Cura #903: Object bounces around in cura

marcvandort, 2016-08-02:
Hello,

I added 3 objects to the build platform. As the second object was moved to make room for the third, it was positioned where object 1 already was. In turn, object 1 was moved into object 3, then back into object 2, etc. etc.

[image]
[image]

I can reproduce the issue by loading the same objects again on an empty build platform. Perhaps it is an idea to move the newly added objects, in stead of the ones already there? This can sometimes be quite annoying, when you've carefully positioned everything and then remember you forgot to add some part.

Ghostkeeper, 2016-08-02:
Agreed. It'd be handier if only the new object moves when loading a new object.

Currently a new object is simply placed in the centre and the push-free functionality then moves objects around to push them free of each other. That raises the question though, what should be done with the push-free functionality when you don't load a new object, but move/scale/rotate an old object? Should only the selected object be moving due to push free?

fieldOfView, 2016-08-02:
On creation (loading, duplicating), the new object should be the one to "find a spot to sit", and the other objects should be left alone if at all possible. When manipulating (moving, scaling, rotating), the other objects should be moving out of the way but the currently selected object should not move by itself.

awhiemstra, 2016-08-02:
We originally had code that would do that at least for the selection. But it got removed. I cannot really remember why but it had something to do with things not properly updating.

marcvandort, 2016-08-02:
I would personally prefer the other objects not to move when scaling/rotating either. If they'd move whilst you're doing that, it would be quite difficult to try to fit pieces tightly together. Why not just show objects in grey if their position is illegal, and leave it up to the user to move which ever part to fix it? In my experience I rarely use the default positions anyways when I'm printing more than 1 piece.

Ghostkeeper, 2017-02-18:
The new product manager for Cura at Ultimaker rejected this issue for being too unimportant. This means that the main developers can't spend time on it.

ChrisTerBeke, 2017-11-14:
Is this still an issue?

Ghostkeeper, 2017-11-23:
Ah, no this was fixed with the new build platform arranger last spring.

closed 2017-11-23 by Ghostkeeper