Ultimaker/Cura #910: [master] Empty/blank text in Menu lists

nickthetait, 2016-08-05:
Application Version:
Master branch of Cura 021a33d6bd5114d3ff7edb8fa0a85c91ba55d08d
Master branch of Uranium 48ec68d98f8fa7af311b0793135ea88d1b7759ac
Platform: Debian Jessie
Qt: 5.6
PyQt: PyQt5_gpl-5.6
Display Driver: Intel Corporation Xeon E3-1200 v2/3rd Gen Core processor Graphics Controller (rev 09) - i915

Steps to Reproduce:
Setup development environment
Run Cura from source code

Actual Results:
Opening a menu entry, many entries are displayed as unselectable or with no text at all. Clicking one of these misbehaving entries does nothing except close the menu. Correctly displayed entries function as expected.

Menus with only broken items: Edit, Preferences, Help
Mixed broken/working menus: File, Settings, Right Click in scene view
Fully working menus: View, Extensions 
[image]
[image]
[image]

Expected results:
Able to use all menu entries

Additional Information:
First reported here: https://github.com/Ultimaker/CuraEngine/issues/375#issuecomment-237258003
Making a new ticket to better figure out root cause and to help people who run into this problem in the future.

Able to load and slice a model fine.

Each time a menu is opened/closed this error message is thrown:
file:///home/aleph/Software/AlephObjects/Cura2/QT/install/5.6/gcc_64/qml/QtQuick/Controls/Private/ScrollViewHelper.qml:137:9: QML Binding: Property 'raised' does not exist on Item.

It appears that the Actions module is not being loaded. However the data contained in QUrl.fromLocalFile(Resources.getPath(CuraApplication.ResourceTypes.QmlFiles, "Actions.qml")) in CuraApplication.py seems correct: PyQt5.QtCore.QUrl('file:///home/aleph/Software/AlephObjects/Cura2/user-interface/cura/../resources/qml/Actions.qml')

Full output in log file https://code.alephobjects.com/file/data/3d6tmoxdb6wbxfru7rrm/PHID-FILE-rqv2zxeantttujdkrwmn/Text_display_issue

I got help from the expert debugger @kakaroto but we are both at a loss on what to investigate at this moment.

thopiekar, 2016-08-07:
Sometimes I'm also facing such strange problems. Of course, newer Cura releases should convert settings, etc. properly on their own, but you can try to clean up all user data.
It probably would make sense to move these directories to a save location, so we can improve the update/upgrade process.

However, during my work on master I'm using this script often:
```
thopiekar@home:~$ cat $(which cura_clean)
#!/bin/bash

rm -rfv ~/.local/share/cura/
rm -rfv ~/.config/cura/
rm -rfv ~/.cache/cura/
```

Ghostkeeper, 2016-08-08:
Might this have something to do with translations? Do you have .mo files in the `resources/i18n` folder?

thopiekar, 2016-08-08:
But shouldn't it fallback to the default translation (English) except
returning a empty text?

Ghostkeeper, 2016-08-08:
Yes, it should, but when's the last time we tested that? :stuck_out_tongue:

thopiekar, 2016-08-08:
Well, I've got German enabled here. Just to see whether the translations
have been updated in [master] or not.

So far I never saw empty texts 🤔

nickthetait, 2016-08-08:
No, I don't have any .mo files

nickthetait, 2016-08-08:
Additionally deleting my config data did not fix it :cry:

nallath, 2016-08-09:
Do you happen to use unity + unbuntu? It kinda looks like the issues we had with that a while ago.

nickthetait, 2016-08-09:
Debian Jessie with XFCE

thopiekar, 2016-08-09:
Well, quickly google'd for the reason
Your PyQt5 version (https://packages.debian.org/search?keywords=pyqt5&searchon=names&suite=stable&section=all) (and probably Qt5, too) is too old.
Many things changed in the last Qt5 and PyQt5 releases, so older versions won't work here.

I saw for a while you are using Debian, so some suggestions:
* Upgrade to Debian Testing or when needed to Experimental
* Create a chroot with Ubuntu Wily or Debian Experimental (if Debian is really needed here for a reason)
* Switch to Ubuntu Xenial and use my [master] PPA. I give "support" in some way, as I need a working environment on my own.

thopiekar, 2016-08-09:
Oops, just overlooked you are using an own installation (?) of a more recent Qt5 and PyQt5 installation. You should make sure that your installed version and the installation provided by APT are not mixed up.

However, I've got a Qt 5.5 + PyQt 5.6 installation here and everything works very well.

kakaroto, 2016-08-09:
Just adding some information here about the issue, if you look at the full log from @nickthetait , you'll notice a lot of `QQmlComponent: Component is not ready` and also these lines :

```
file:///home/aleph/Software/AlephObjects/Cura2/user-interface/resources/qml/Cura.qml:72:29: Unable to assign [undefined] to QQuickAction*
file:///home/aleph/Software/AlephObjects/Cura2/user-interface/resources/qml/Cura.qml:107:36: Unable to assign [undefined] to QQuickAction*
file:///home/aleph/Software/AlephObjects/Cura2/user-interface/resources/qml/Cura.qml:111:36: Unable to assign [undefined] to QQuickAction*
file:///home/aleph/Software/AlephObjects/Cura2/user-interface/resources/qml/Cura.qml:118:36: Unable to assign [undefined] to QQuickAction*
file:///home/aleph/Software/AlephObjects/Cura2/user-interface/resources/qml/Cura.qml:119:36: Unable to assign [undefined] to QQuickAction*
file:///home/aleph/Software/AlephObjects/Cura2/user-interface/resources/qml/Cura.qml:121:36: Unable to assign [undefined] to QQuickAction*
file:///home/aleph/Software/AlephObjects/Cura2/user-interface/resources/qml/Cura.qml:122:36: Unable to assign [undefined] to QQuickAction*
file:///home/aleph/Software/AlephObjects/Cura2/user-interface/resources/qml/Cura.qml:123:36: Unable to assign [undefined] to QQuickAction*
```

All of these `[undefined]` warnings are matching all the lines in which `Cura.Actions.<property>` is used. Also worth nothing, all the empty menu items are the `Cura.Actions.<property>` items, while the menu items that appear correctly are not referencing `Cura.Actions`. I've asked @nickthetait to print the value of Cura.Actions and it does get loaded (it's not undefined itself), however, it doesn't have any valid properties, which is why the menu can't reference any of the actions (== empty item). 

I have also tried some things and was able to reproduce the error, more specifically, if I remove the 'pragma singleton' from the Actions.qml file, then I get the same behavior and warnings as @nickthetait , which led me to believe that it's possible that something is causing the Cura.Actions module not to be loaded as a singleton, but I was unable to figure out what exactly would cause it to be loaded as a regular type instead of a singleton.
I hope this helps...

Ghostkeeper, 2016-08-10:
We require either Qt 5.6 or later, or Qt 5.4 or later with a patch to fix Qt's garbage collection deadlock (found [here](https://github.com/Ultimaker/cura-build/blob/master/qpyqmlobject.h.patch) and [here](https://github.com/Ultimaker/cura-build/blob/master/qpyqmlobject.cpp.patch).

thopiekar, 2016-08-10:
I can confirm PyQt 5.6 and Qt 5.5 works here, too. The patches @Ghostkeeper mentions are actually for PyQt5 as you can see here: https://github.com/Ultimaker/cura-build/blob/master/CMakeLists.txt#L289

nickthetait, 2016-08-10:
Just tried building and installing, the menus look fine via that method. 
[image]

So this problem is definitely specific to how a development environment is setup.

thopiekar, 2016-08-10:
Cura always has dependencies on very latest releases, but the distributions doesn't come with them that fast. I'm noticing a kind freeze here since last week and I'm sure it's because of my Qt5.5 installation.

Ghostkeeper, 2016-08-11:
If that freeze happens most often when you close windows, switch tabs or profiles, or close the tool panels, then it's probably the deadlock that was being fixed by that patch.
We haven't noticed any UI freezes recently here on our side.

nickthetait, 2016-08-22:
Fixed by PR 940

nickthetait, 2016-08-23:
Back to basics... so are these development environment setup instructions sane? https://code.alephobjects.com/P13

nallath, 2016-08-24:
You also need numpy-stl (can get it through sip). Cura will run fine without it, but STl loading will be way faster. 

Scypi is definately needed! We use it for the convex hull calculations.

thopiekar, 2016-08-24:
@nallath I'm sure you meant pip not sip 😜

nallath, 2016-08-24:
Err. Yes.

closed 2016-09-18 by thopiekar