Ultimaker/Cura #911: Feature Request: Z Hop Always On (or Z Hop when crossing infills)

innovo3d, 2016-08-06:
[image]

I'm trying to use Cura + my printer as a plotter (blade to cut vinyl or pen to draw) and this could work perfectly with Z hops, no bottom infill and printing perimeters only except that when you have to gap inside of an object (for example tracing an O), Cura does not perform the Z hop between the outside perimeter and the inside perimeter because it assumes that it's an infill and Z hops don't matter here

With 3D printing this is perfectly fine since any plastic extruded or scars created in between would be filled over by subsequent layers but for plotter applications this means big unwanted lines across the artwork.

I would like to request to check **Z hop Always On** for these kinds of applications.

This could alternatively be a check for **Enable Z hops when crossing infills**.  
This would be particularly helpful for the 'Z hops on Top Layer Only' feature request also because it can help avoid the nozzle from dragging across infills and scarring prints.

Ghostkeeper, 2016-08-08:
**Enable Z hops when crossing infills** or something like that would be better, I think.

As a workaround, for now, you could make your meshes hollow. If you use a wall thickness between one line width and two line widths, it'll only draw one line.

By the way, you can turn off the skirt by selecting brim and a brim width of 0. Perhaps that's something you would like to do in your case :wink:

innovo3d, 2016-08-08:
Thanks.

By meshes you mean infill, correct? That is what I did; single perimeter,
no bottom infill, 0% infill. The problem is that Cura still thinks that
there is an infill when there exists another perimeter inside of an object.

For example, imagine tracing the letter "O". First it traces the outside
perimeter of the O, then it traces the inside perimeter of the O. When it
crosses from outside to inside it still assumes that this region is an
infill so it will not perform Z hops. Sadly even choosing 0% infill does
not make Cura realize that there is no infill there.

Ghostkeeper, 2016-08-08:
No, what I meant is to make the actual meshes hollow, so it doesn't think that space is infill any more :stuck_out_tongue:
Not really a workaround, I know...

DDDirk, 2016-08-08:
Clarification: the mesh is the collection of vertices and lines that connect them, which define the shape of the model.

To make Cura accept that it is a hollow model, you may have to disable Mesh Fixes > Union Overlapping Volumes.

crozone, 2020-01-18:
Why was this closed?

Ghostkeeper, 2020-01-21:
It shouldn't have been. It's a valid issue, though be aware that it has no priority for us to implement so I don't expect we'll be able to implement this any time soon. For most prints it'll be detrimental for both quality and productivity, is my guess.

crozone, 2020-01-22:
> be aware that it has no priority for us to implement so I don't expect we'll be able to implement this any time soon

Understood. I ask because for whatever reason, CURA doesn't always perform retractions on some large moves, and the lack of Z-hop over large surfaces causes scraping and scars on the top layers of prints.

On machines with an extremely fast Z axis (like Deltabots which can hit ~300mm/s or faster), the quality is not degraded significantly because any oozing plastic is "snapped" instead of drawn out of the nozzle by the hop. The print time is also unaffected because the z-hop is so fast that it only makes up a fraction of the move duration.

To reduce implementation complexity, even the ability to "z-hop on moves longer than X distance" without any infill detection would still be a great feature. I will try to implement a proof of concept as a g-code post processing script to gauge print quality deltas.

GregValiant, 2025-07-03:
This is 9 years old so I'll go ahead and close it.