Ultimaker/Cura #913: No renderer from windows object to print

sappas, 2016-08-08:
Hello, I've no renderer from the objet to print.

Please find follow, cura --debug 

Version: 2.1.2-0
Platform: Linux-4.4.0-31-generic-x86_64-with-Ubuntu-16.04-xenial
Qt: 5.5.1
PyQt: 5.5.1

Exception:
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/UM/Qt/Bindings/MainWindow.py", line 172, in _render
    renderer.beginRendering()
  File "/usr/lib/python3/dist-packages/UM/Qt/QtRenderer.py", line 104, in beginRendering
    self._initialize()
  File "/usr/lib/python3/dist-packages/UM/Qt/QtRenderer.py", line 158, in _initialize
    OpenGL.setInstance(QtOpenGL())
  File "/usr/lib/python3/dist-packages/UM/Qt/GL/QtOpenGL.py", line 25, in __init__
    QMessageBox.critical("Failed to Initialize OpenGL", "Could not initialize OpenGL. This program requires OpenGL 2.0 or higher. Please check your video card drivers.")
TypeError: QMessageBox.critical(QWidget, str, str, QMessageBox.StandardButtons buttons=QMessageBox.Ok, QMessageBox.StandardButton defaultButton=QMessageBox.NoButton): argument 1 has unexpected type 'str'

thopiekar, 2016-08-08:
The exceptions says everything: ```QMessageBox.critical("Failed to Initialize OpenGL", "Could not initialize OpenGL. This program requires OpenGL 2.0 or higher. Please check your video card drivers.")```

It looks like you are using Windows' basic graphics driver.

Ghostkeeper, 2016-08-08:
> It looks like you are using Windows' basic graphics driver.

He's on Ubuntu 16.04! But you're right, it's the graphics driver that is not being found or not up to date.

sappas, 2016-08-08:
Thanks a lot, I've 945GM Intel card graphics who is limited to OpenGL 1.4 !
I've to change my laptop, use Cura 15.04 or other slicer.
You can close this issue.

thopiekar, 2016-08-08:
No problem. I also (hopefully) fixed the issue which is mentioned in the logs.
In Cura's next release you should get a window getting up, that your drivers are not working well.

Would be great if you could bookmark this issue and give feedback on Cura's next release if you can.

Thanks!

closed 2016-08-08 by thopiekar

sappas, 2016-08-08:
Ok, I'll check it,