Ultimaker/Cura #915: Cura 2.1.2 - Creating a profile with a new name triggers a re-slice

nicolinux, 2016-08-10:
After I have set the slicing settings, I want to save it as a new profile (Menu: Profile -> Add Profile). When I rename the profile (because I can't have it named "Customized #2"), Cura starts the slicing again. This is not necessary.

Ghostkeeper, 2016-08-11:
Indeed it does, also in the latest master version.

I think it happens because it "switches profiles".

Fixing this we need to take caution to what happens to the profile that's being put at the end of the g-code. Though iirc it does get appended to the g-code upon saving the g-code, not directly after slicing, so it should be possible.

Ghostkeeper, 2016-08-11:
This was just scheduled for fixing in version 2.3.

thopiekar, 2016-09-17:
Fix already released (CURA-2109)

closed 2016-09-17 by thopiekar