Ultimaker/Cura #917: Center of object differs when printing + object scaled smaller

filamentworld, 2016-08-10:
Based on Cura 2.1.2

In preparing to generate Gcode, I always make sure the object is centered on the print bed. I now found out that when printing the object, the center of the physically printed object has shifted to the left.
Also I noticed that the object when printed, has become smaller than the digital version. I think about 5-10%.

Ghostkeeper, 2016-08-11:
The centre of the physical bed is not necessarily the centre of the build volume. The movement of the print head in Ultimaker printers is slightly limited on one side due to the asymmetric placement of the nozzle. It might also shift if you haven't put your glass plate exactly in the centre. For different printers there may be other causes for asymmetry.

We've done quite extensive tests to make sure that the printed object has the correct dimension, but of course some materials shrink a lot when cooling.
Are you certain that Cura didn't scale the object automatically to make it fit in the build volume?

nallath, 2017-07-27:
Closing due to inactivity.

closed 2017-07-27 by nallath