Ultimaker/Cura #919: Cura 2.1.2 - Allow to change to regular RepRap Gcode

nicolinux, 2016-08-11:
Right now you have to edit the .json definitions which is super user unfriendly (if you have selected an Ultimaker 2 printer). However there are valid use cases where an Ultimaker 2 user wants to have regular gcode and not Ultigcode.

fieldOfView, 2016-08-11:
See https://github.com/fieldofview/machinesettingsactionplugin. We plan to include this plugin in Cura 2.2. It currently looks like this:
[image]

fieldOfView, 2016-08-11:
For 2.1, you can add this machine definition to cura/resources/definitions:
https://gist.github.com/fieldOfView/ae85f755fc70d1c65e6968a870dfef0c

closed 2016-08-11 by fieldOfView

nicolinux, 2016-08-11:
Perfect, thanks!