Ultimaker/Cura #925: Default Temperature not reset for PLA - Normal Quality

innovo3d, 2016-08-16:
When you first start Cura (I'm using the Innovo Inventor profile), the default filament is ABS, which has a print temperature of 250 degrees (which I find extremely high, no? 230 should be enough!).  When I change this to PLA, it updates the settings for the bed (changes from 80 to 60) but still keeps the extruder temperature at 250 degrees!

For my printer, this is disastrous if you already have PLA loaded in the hot end, as it will expand and jam inside of the extruder.

I'm pretty sure this is a bug, since the heated bed temperature DID change, but the extruder temperature didn't.

Also, is there a way to set default print temperatures through the overrides in the printer profiles?  I'd like to set the default extrude temperature to 205 and default bed temp to 0.  I've already set this in my printer profile JSON file, but it doesn't seem to have any affect on the settings I'm seeing in Cura.

Overrides:
"material_print_temperature": { "default": 205, "visible": true },
"material_bed_temperature": { "default": 0, "visible": true }

fieldOfView, 2016-08-16:
Yes, this is a buggy part of Cura 2.1.x (because I wrote it? :-/ ). It has been much improved in Cura 2.2 (the current master branch).

I think you may have to edit the values in resources/profiles/materials. They were not meant to override your own settings, but apparently they do. You could also have a look at how resources/machines/ultimaker.json has a "preferred material" set to PLA and add that to your own printer profile for now.

innovo3d, 2016-08-16:
Thanks. Is there also a way to set my own temperature for PLA or my own
material profiles?

fieldOfView, 2016-08-16:
AFAIK only by editing the files in resources/profiles/materials, due to the aforementioned bug.

innovo3d, 2016-08-16:
I checked the Ultimaker JSON file and didn't see the preferred material
selection anywhere in there.  I just pulled the latest file and still don't
see it on there.   I checked the inherited fdmprinter.json file as well and
didn't see it, but maybe I didn't look carefully enough.  Let me know if
it's somewhere else.

Thanks,

Adam

*Adam Rumjahn*

* <http://www.innovo3d.co>*

*adam@innovo3d.co <adam@innovo3d.co>Cel: +57 304 643 1248*
*Fijo: +57 2 374 2473*

innovo3d, 2016-08-16:
By the way, I think I found the problem:

*in the ABS.cfg file you define the material print temp:*
[general]
version = 1
type = material
name = ABS

[settings]
*material_print_temperature = 250*
material_bed_temperature = 80
material_flow = 107
retraction_amount = 6.5

*However, in the PLA.cfg file you don't:*
[general]
version = 1
type = material
name = PLA

[settings]
material_bed_temperature = 60
retraction_amount = 6.5

That's probably why when you switch over to PLA it keeps the temperature at
250.  But where does the 215 come from? I don't see it anywhere.

*Adam Rumjahn*

* <http://www.innovo3d.co>*

*adam@innovo3d.co <adam@innovo3d.co>Cel: +57 304 643 1248*
*Fijo: +57 2 374 2473*

fieldOfView, 2016-08-16:
Prefered material, see:
https://github.com/Ultimaker/Cura/blob/2.1/resources/machines/ultimaker.json#L10

If you add a print_temperature to the PLA file that should indeed fix this issue.

innovo3d, 2016-08-16:
> machines/ultimaker.json#L10

Ahhh! you meant the Ultimaker Original.  All this time I was checking
Ultimaker2, 2+, extended! haha.  Ok, gotcha!  Thanks.

Sending a new pull request this afternoon.

*Adam Rumjahn*

* <http://www.innovo3d.co>*

*adam@innovo3d.co <adam@innovo3d.co>Cel: +57 304 643 1248*
*Fijo: +57 2 374 2473*

fieldOfView, 2016-08-16:
No, actually all the Ultimaker printers use an extra layer of inheritance to make some common settings.

fieldOfView, 2016-08-16:
Closing this issue because it is known and it has been fixed in 2.2

closed 2016-08-16 by fieldOfView

innovo3d, 2016-08-16:
Thanks, Aldo, one last question.

I'm trying to test out my JSON file, but it looks like as soon as I make
modifications to it it becomes NULL in Cura 2.1.2 and I can't select my
printer anymore.  Is there any workaround for this so that I can test out
some of my settings before merging?  It would be really helpful.

Thanks,

*Adam Rumjahn*

* <http://www.innovo3d.co>*

*adam@innovo3d.co <adam@innovo3d.co>Cel: +57 304 643 1248*
*Fijo: +57 2 374 2473*

innovo3d, 2016-08-16:
Nevermind.  I found the error.  It was in the Machine Preferences code that
I added.  Damn brackets and commas!

*Adam Rumjahn*

* <http://www.innovo3d.co>*

*adam@innovo3d.co <adam@innovo3d.co>Cel: +57 304 643 1248*
*Fijo: +57 2 374 2473*

Ghostkeeper, 2016-08-16:
JSON comma's get me every time too, @innovo3d :smile:

innovo3d, 2016-08-17:
And those damn spaces in Python! haha

*Adam Rumjahn*

* <http://www.innovo3d.co>*

*adam@innovo3d.co <adam@innovo3d.co>Cel: +57 304 643 1248*
*Fijo: +57 2 374 2473*