Ultimaker/Cura #927: segfault in code built with cura-build.

kenjo, 2016-08-17:
So I built from what I guess is git master for yesterday. here is the git commit id for the different packages used. 
```
5f8ad4a2a6a9803bf7597a883787a9aa5128cc6c  /mnt/data/cura-build/build/Cura-PostProcessing-prefix/src/Cura-PostProcessing
12697230b300a9e51f77753f35198d2597eea018  /mnt/data/cura-build/build/Arcus-prefix/src/Arcus
063faf1520e1b799cb34f302d4d4c6b38d20bf89  /mnt/data/cura-build/build/Cura-prefix/src/Cura
aa05d483b6f9a958c4b66b766f42769660c5ebc4  /mnt/data/cura-build/build/cura-binary-data-prefix/src/cura-binary-data
6db043a0c14393d74dcb25e9e90eab3c15afd5c1  /mnt/data/cura-build/build/Cura-Doodle3D-prefix/src/Cura-Doodle3D
7d158cbce2a1ec95fb4aab2fa51f6ac2a88a155a  /mnt/data/cura-build/build/Uranium-prefix/src/Uranium
bf0cd02928dac658ead2a61a7e2c71547880d6b2  /mnt/data/cura-build/build/CuraEngine-prefix/src/CuraEngine

```

problem is that I end up in a segmentation fault before the program really starts.

```

gdb /usr/bin/python3.5 -c CoreDump
GNU gdb (Ubuntu 7.11.1-0ubuntu1) 7.11.1
Copyright (C) 2016 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.  Type "show copying"
and "show warranty" for details.
This GDB was configured as "x86_64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<http://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
<http://www.gnu.org/software/gdb/documentation/>.
For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from /usr/bin/python3.5...Reading symbols from /usr/lib/debug/.build-id/58/5d2e9acfcdbb49ae6f6d3d19990cfe149c6781.debug...done.
done.
[New LWP 30566]
[New LWP 30583]
[New LWP 30579]
[New LWP 30588]
[New LWP 30589]
[New LWP 30578]
[New LWP 30582]
[New LWP 30584]
[New LWP 30586]
[New LWP 30596]
[New LWP 30580]
[New LWP 30593]
[New LWP 30590]
[New LWP 30594]
[New LWP 30575]
[New LWP 30581]
[New LWP 30585]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/x86_64-linux-gnu/libthread_db.so.1".
Core was generated by `python3.5 /opt/cura/bin/cura'.
Program terminated with signal SIGSEGV, Segmentation fault.
#0  0x00007f0bdf313e3f in ?? () from /opt/cura/lib/libQt5Qml.so.5
[Current thread is 1 (Thread 0x7f0bfe598700 (LWP 30566))]
(gdb) bt
#0  0x00007f0bdf313e3f in ?? () from /opt/cura/lib/libQt5Qml.so.5
#1  0x00007f0bdf2d0631 in QV4::ExecutionEngine::ExecutionEngine(QV4::EvalISelFactory*) () from /opt/cura/lib/libQt5Qml.so.5
#2  0x00007f0bdf414683 in QV8Engine::QV8Engine(QJSEngine*) () from /opt/cura/lib/libQt5Qml.so.5
#3  0x00007f0bdf25ebb4 in QJSEngine::QJSEngine(QJSEnginePrivate&, QObject*) () from /opt/cura/lib/libQt5Qml.so.5
#4  0x00007f0bdf37f219 in QQmlEngine::QQmlEngine(QQmlEnginePrivate&, QObject*) () from /opt/cura/lib/libQt5Qml.so.5
#5  0x00007f0bdf4004e0 in QQmlApplicationEngine::QQmlApplicationEngine(QObject*) () from /opt/cura/lib/libQt5Qml.so.5
#6  0x00007f0bdf5ab009 in sipQQmlApplicationEngine::sipQQmlApplicationEngine(QObject*) () from /opt/cura/lib/python3/dist-packages/PyQt5/QtQml.so
#7  0x00007f0bdf5ab2ea in init_type_QQmlApplicationEngine () from /opt/cura/lib/python3/dist-packages/PyQt5/QtQml.so
#8  0x00007f0be1bca18b in ?? () from /opt/cura/lib/python3/dist-packages/sip.so
#9  0x000055de00aa2944 in type_call.lto_priv () at ../Objects/typeobject.c:905
#10 0x000055de00ae0057 in PyObject_Call () at ../Objects/abstract.c:2166
#11 0x000055de00a49111 in do_call (nk=<optimized out>, na=0, pp_stack=0x7ffce0dd1e30, func=<optimized out>) at ../Python/ceval.c:4936
#12 call_function (oparg=<optimized out>, pp_stack=0x7ffce0dd1e30) at ../Python/ceval.c:4732
#13 PyEval_EvalFrameEx () at ../Python/ceval.c:3236
#14 0x000055de00a48b4f in fast_function (nk=<optimized out>, na=<optimized out>, n=<optimized out>, pp_stack=0x7ffce0dd1f60, func=<optimized out>) at ../Python/ceval.c:4803
#15 call_function (oparg=<optimized out>, pp_stack=0x7ffce0dd1f60) at ../Python/ceval.c:4730
#16 PyEval_EvalFrameEx () at ../Python/ceval.c:3236
#17 0x000055de00a48b4f in fast_function (nk=<optimized out>, na=<optimized out>, n=<optimized out>, pp_stack=0x7ffce0dd2090, func=<optimized out>) at ../Python/ceval.c:4803
#18 call_function (oparg=<optimized out>, pp_stack=0x7ffce0dd2090) at ../Python/ceval.c:4730
#19 PyEval_EvalFrameEx () at ../Python/ceval.c:3236
#20 0x000055de00a4d4f6 in _PyEval_EvalCodeWithName () at ../Python/ceval.c:4018
#21 0x000055de00a4e21f in PyEval_EvalCodeEx () at ../Python/ceval.c:4039
#22 PyEval_EvalCode (co=<optimized out>, globals=<optimized out>, locals=<optimized out>) at ../Python/ceval.c:777
#23 0x000055de00b26892 in run_mod () at ../Python/pythonrun.c:976
#24 0x000055de00b28dbd in PyRun_FileExFlags () at ../Python/pythonrun.c:929
#25 0x000055de00b295be in PyRun_SimpleFileExFlags () at ../Python/pythonrun.c:396
#26 0x000055de00b59b27 in run_file (p_cf=0x7ffce0dd2300, filename=0x55de014ba1a0 L"/opt/cura/bin/cura", fp=0x55de015b2e80) at ../Modules/main.c:318
#27 Py_Main () at ../Modules/main.c:768
#28 0x000055de009ea451 in main () at ../Programs/python.c:65
#29 0x00007f0bfdde77e0 in __libc_start_main (main=0x55de009ea370 <main>, argc=2, argv=0x7ffce0dd2518, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7ffce0dd2508)
    at ../csu/libc-start.c:291
#30 0x000055de00af0119 in _start ()
(gdb) 

```

not sure how to get any info out from the actual python program.

kenjo, 2016-08-17:
I found out how to get a backtrace for python.

```
(gdb) py-bt
Traceback (most recent call first):
  File "/opt/cura/lib/python3/dist-packages/UM/Qt/QtApplication.py", line 135, in initializeEngine
    self._engine = QQmlApplicationEngine()
  File "/opt/cura/lib/python3/dist-packages/cura/CuraApplication.py", line 424, in run
    self.initializeEngine()
  File "/opt/cura/bin/cura", line 62, in <module>
    app.run()

#13 Frame 0x7f0b965923d8, for file /opt/cura/lib/python3/dist-packages/UM/Qt/QtApplication.py, line 135, in initializeEngine (self=<CuraApplication(visibleMessageAdded=<Signal(_Signal__signals=<WeakImmutableList(_WeakImmutableList__list=[]) at remote 0x7f0bdce57518>, _Signal__methods=<WeakImmutablePairList(_WeakImmutablePairList__list=[]) at remote 0x7f0bdce574e0>, _Signal__functions=<WeakImmutableList(_WeakImmutableList__list=[]) at remote 0x7f0bdce574a8>, _Signal__lock=<_thread.lock at remote 0x7f0be17062d8>, _Signal__type=2) at remote 0x7f0bdce57470>, _splash=<CuraSplashScreen(_scale=1) at remote 0x7f0bdce5d1f8>, _required_plugins=['CuraEngineBackend', 'MeshView', 'LayerView', 'STLReader', 'SelectionTool', 'CameraTool', 'GCodeWriter', 'LocalFileOutputDevice'], _physics=<PlatformPhysics(_build_volume=<BuildVolume(boundingBoxChanged=<Signal(_Signal__signals=<WeakImmutableList(_WeakImmutableList__list=[]) at remote 0x7f0b96692cc0>, _Signal__methods=<WeakImmutablePairList(_WeakImmutablePairList__list=[(<weakref at...(truncated)
    self._engine = QQmlApplicationEngine()
#16 Frame 0x55de03e90e78, for file /opt/cura/lib/python3/dist-packages/cura/CuraApplication.py, line 424, in run (self=<CuraApplication(visibleMessageAdded=<Signal(_Signal__signals=<WeakImmutableList(_WeakImmutableList__list=[]) at remote 0x7f0bdce57518>, _Signal__methods=<WeakImmutablePairList(_WeakImmutablePairList__list=[]) at remote 0x7f0bdce574e0>, _Signal__functions=<WeakImmutableList(_WeakImmutableList__list=[]) at remote 0x7f0bdce574a8>, _Signal__lock=<_thread.lock at remote 0x7f0be17062d8>, _Signal__type=2) at remote 0x7f0bdce57470>, _splash=<CuraSplashScreen(_scale=1) at remote 0x7f0bdce5d1f8>, _required_plugins=['CuraEngineBackend', 'MeshView', 'LayerView', 'STLReader', 'SelectionTool', 'CameraTool', 'GCodeWriter', 'LocalFileOutputDevice'], _physics=<PlatformPhysics(_build_volume=<BuildVolume(boundingBoxChanged=<Signal(_Signal__signals=<WeakImmutableList(_WeakImmutableList__list=[]) at remote 0x7f0b96692cc0>, _Signal__methods=<WeakImmutablePairList(_WeakImmutablePairList__list=[(<weakref at remote 0x7f...(truncated)
    self.initializeEngine()
#19 Frame 0x7f0bfe54f828, for file /opt/cura/bin/cura, line 62, in <module> ()
    app.run()

```

thopiekar, 2016-08-17:
Other people and me got the same problem. I didn't know that gdb comes with that tool.
The output looks very useful by the way. Got no experiences which info is really relevant and helping, but I'm sure someone of the others can give feedback on this.

Thanks!

Ghostkeeper, 2016-08-18:
Well, what this tells me is that you're getting a segfault in your Qt installation. My guess is that it is due to PyQt, its interface with Qt, or its interface with Python, rather than inside Qt itself (but you never know, since Qt has about [8.000 issues](https://bugreports.qt.io/browse/QTWEBSITE-723?jql=status%20in%20(Open%2C%20%22In%20Progress%22%2C%20Reopened%2C%20%22To%20Do%22)) open in their issue tracker). The code in those stack frames you posted hasn't changed in a very long while, and we haven't seen any problems with it yet.

Perhaps your PyQt picked up a different installation of Qt from your system? Perhaps your PyQt isn't compiled with the same compiler as what your Python installation was compiled with?

kenjo, 2016-08-18:
I use the cura-build project, is not the purpose of that project to do a correct build of cura ? if it fails there is a problem. now where that problem actually is located I don't really know but the fact that the seqfault is in /opt/cura/lib/libQt5Qml.so.5 don't mean the problem is actually there it could be some structure that pyhton pass in to the qt libraries. In this case it looks like it fails really early so something fundamental looks to be wrong. 

I noticed that there is some issue with cmake not finding the correct path to header files and libs for arcus and maybe something like that is happening also for pyqt/qt.

thopiekar, 2016-09-17:
Hey @kenjo,
remember you mailed me because of my PPA recently.
Do you have problems with that issue?

Thanks!

nallath, 2017-07-27:
Closing due to inactivity.

closed 2017-07-27 by nallath