Ultimaker/Cura #931: Stuck at prepare Slice

TYR3D, 2016-08-18:
Once object was dropped in. It prepared the slice but when I rotate any way it then just says preparing slice but nothing happens

Ghostkeeper, 2016-08-19:
In 2.1, there is a known bug that it gets stuck in Preparing to Slice sometimes when you use one-at-a-time mode.

In the current master, it seems to get stuck in preparing to slice sometimes if you interrupt the slicing process while it is already preparing to slice.

TYR3D, 2016-08-19:
Hi just to let you know it happen in all at once mode too.
Thanks

nallath, 2016-08-19:
I'm working on this issue as we speak. It's a nasty one, as it also behaves differently on different operating systems.

TYR3D, 2016-08-19:
when placing the item on the bed and rotate it some times it bounces all over the place and disappears as well need to then turn of and reboot. Might have something to do with that too?

Ghostkeeper, 2016-08-19:
Happens a lot with printing bouncy balls out of flex materials, I've heard :laughing:

No but the bouncing is also a known issue. It seems to happen more with combinations of large and small objects on the build plate, but we haven't really analysed it yet.

nallath, 2016-08-19:
Yes we did. I just made a fix for that :) Not sure if it solves all cases though.

robygladio, 2016-08-24:
[SOLVED] On WINDOWS 10 64 bit just execute in compatibility mode windows 8 
I flag also Run As Administrator for both cura and curaengine.exe
Now work again

sedwards2009, 2016-08-24:
@robygladio Does it work now using compatibility mode but *not* using admin rights?

robygladio, 2016-08-26:
@sedwards2009   
I use admin rights  too. 
both cura and curaengine.exe set as compatibility win 8 and admin rights.
Just the dual extrusion get crash if set as 1. leaving dual extrusion to 0 it work fine

mokusolo, 2016-08-31:
I had the similar problem when using 2.1.2 with win8. To test,the stl file size is nearly 57M.
I would appreciate that if anyone can tell me how long the normal time it costs when"preparing to slice".
Does it relate to file size/transmission speed from cura to curaEngine(I'm not sure if there's transmission)?
Thanks a lot anyway.

fieldOfView, 2016-08-31:
The "preparing to slice" stage should not take too long. For small objects you should not really see it much if at all. Highly detailed objects may take longer, but I would say that if it takes more than tens of seconds, something is wrong.

Important question: is the model yellow or grey? If it is grey, Cura thinks it does not fit in the buildvolume and does not slice it.

Ghostkeeper, 2016-09-04:
Preparing to slice is indeed 99% the transmission towards CuraEngine. There's a few other checks such as which meshes to slice (only those in the build volume), what order to print in one-at-a-time mode, and a few tiny things.

Cura 2.1 had a bug where it would sometimes still try to slice, and get stuck in preparing to slice, even if the model was grey.

thopiekar, 2016-09-17:
@TYR3D Is that problem still existing on Cura 2.3?

TYR3D, 2016-09-18:
Seams to be gone. Found my self now using other slicers a little more as it is still too unstable

closed 2016-09-19 by nallath