Ultimaker/Cura #932: Cura 2.1.3 - Creating a profile with a new name triggers a re-slice

QoooQ, 2016-08-19:
After I have set the slicing settings, I want to save it as a new profile (Menu: Profile -> Add Profile). When I rename the profile (because I can't have it named "Customized #2"), Cura starts the slicing again. This is not necessary.

2.1.2 have this bug       but 2.1.3 not fixing it?!
(Retraction Distance)  is 6.5mm  can not customized (EX.3mm)   Why?!

nallath, 2016-08-19:
Because if we don't know about a bug, we can't fix it.

Retraction distance can be changed in my version (master, the soon to be 2.2)

Ghostkeeper, 2016-08-19:
We do know about this bug, actually (CURA-2109 in our internal tracker).

Retraction distance can be changed, except if you select any of the the Ultimaker 2 family of printers. Those printers don't allow for changing the retraction distance via g-code, so Cura can't set it.

thopiekar, 2016-09-17:
Fix already released

closed 2016-09-17 by thopiekar