Ultimaker/Cura #933: Win10 anniversary update - Crash on application start 2.1.3

surfaace, 2016-08-19:
After annaversay update of 64 bit win10 cura crashes on start with reference to opengl32.dll
Crash results is only found from evenviewer attached below. Before updating cura was launched and slicing worked :+1: 

Worth mentioning is that that same applicationcrash is caused with 32 and 64 bit version of 2.1.3. Legacy 15.x.x can be started, but is not slicing which slicing issue has been reported.

[cura crash.zip](https://github.com/Ultimaker/Cura/files/426941/cura.crash.zip)

mebrein, 2016-09-04:
Same syptoms here on a Lenovo W550s with NVIDIA Quadro K620M with Anniversy Update. 

When starting Cura the following is show:
- Loading plugins
- Loading machines
- Loading interface
crash.

The legacy 15.x.x start, but is not slicing too. So same symptoms over here.

When running on a laptop without the NVIDIA chipset no problems whatsoever.

sedwards2009, 2016-09-04:
It is first worth updating your video drivers. This appears to be driver related.

mebrein, 2016-09-04:
Well, in fact I already did that. Even fully removed all NVIDIA driver, in de device driver. Tried both drivers from windows 10. Looked whether Cura was ok. After that removed again and manually installed from NVIDIA website. Still same problem. Have no clue, just reporting so you guys might have some input because I see this bug is a release blocker.

closed 2018-02-21 by ianpaschal