Ultimaker/Cura #934: Resize/Scale giving incorrect percentage

8biten, 2016-08-19:
When I try to resize a file I created in Pro/E - it gives incorrect percentages, such as 700 or 1000+ on an axis.  Upon exiting, the object is huge.  Doing an undo/ctl-Z will then bring the object to the correct (intended size after resize) size.  
[image]
[stl_file.zip](https://github.com/Ultimaker/Cura/files/427704/stl_file.zip)

nallath, 2016-08-22:
We re-did a bunch of code that has to with scaling for 2.2, which will be released soon.

Ghostkeeper, 2016-08-22:
You're doing a uniform rescale?

The percentages look fine in comparison with the dimensions it reports. The mesh is only 4mm wide in the Y-direction. If it's not doing a uniform rescale while you had the uniform scaling activated, then we probably already fixed that just after 2.1 was released when we fixed a lot of things with scaling.

8biten, 2016-08-23:
Hmm, I did rotate it first.  It is not 4mm wide - it is 4mm tall.  (it was on it's side in the STL file).

nallath, 2017-07-27:
This is probably fixed already.

closed 2017-07-27 by nallath