Ultimaker/Cura #935: Cura-2.1.3-Linux.deb is not complete

bircoe, 2016-08-20:
I've tried downloading the Cura-2.1.3-Linux.deb file from multiple connections and devices, result is always the same, partial file, it seems there is an incomplete file on the web server, the 2.1.3 deb is one third the size of the 2.1.2 deb:

[image]

closed 2016-08-20 by bircoe

fieldOfView, 2016-08-20:
@bircoe, did you mean to close this issue? Is the problem solved?

bircoe, 2016-08-20:
I did mean to close it, it's sort of resolved. The above screenshot is from the "current" directory, the 2.1.3 deb file is complete in the "2.1.3-pre-release" directory.
So this one is incomplete:
http://software.ultimaker.com/current/Cura-2.1.3-Linux.deb
This one is complete:
http://software.ultimaker.com/2.1.3-pre-release/Cura-2.1.3-Linux.deb