Ultimaker/Cura #937: Cura does not start slicing, and there's no print timer displayed.

laban93, 2016-08-21:
The cura program does freeze up quite often, so i need to exit and relaunch the program. Doing that several times before it recognises the model and starts slicing. The print timer is also set to 00h 00m.

Ghostkeeper, 2016-08-22:
Stuck in "Preparing to slice..."?

robygladio, 2016-08-23:
Same problem there.
before I can slice without problem now is an infinite preparing for slicing..
curaengine is set as RUN AS ADMINISTRATOR, no firewall

Ghostkeeper, 2016-08-23:
That's a duplicate of #931.

3DKernow, 2016-08-30:
I'm having the same problem with 2.1.3. Some stls slice OK. Others don't.

3DKernow, 2016-08-30:
I've found the answer for my problem from a comment in another discussion strand, If you have set the software to support printing one at a time (which I had done), then the software has problems with designs above a certain height, because (I assume) the software needs to stop the nozzle bumping into a finished object. I changed the setting and my problem stl immediately sliced.

3DKernow, 2016-08-30:
.... and I had another stl which was giving slicing problems. This was a design I had pulled off Thingiverse and it had downloaded as a very small object (11*17*5 mm), Once I scaled it up, it started slicing fine. Just doubling the size did the trick.
CONCLUSIONS:  slicing is sensitive when files are scaled down too much. There is also at least one setting (print one at a time) which affects files which are taller than a certain height.

awhiemstra, 2016-09-05:
The error reporting about build volume height should be better in 2.3, we made several fixes to that. Similarly, we made fixes to the auto scaling to scale very small models up. So I think both issues should be resolved in 2.3.

closed 2016-09-05 by awhiemstra