Ultimaker/Cura #938: On cura2.1 running pyqt5 deadlock problem!

twosilly-2B, 2016-08-22:
Hello! I see your code on GitHub! It's great! It's a great setting!
I can't wait to learn from him, but when I use the code to run him up and find bug!
I don't know whether my setting is wrong or something!
When I want to set visibility setting the program has no reason to crash!
I have tried for a long time can not find the reason!
I hope you can solve my problem, thank you! Compiler environment (windows7-64 bit python3.4 pyqt5.5)
bug

[image]

It's here! When I slide down to set him up! The program is crashing! I really can't find the reason!

The facts proved to be pyqt5's mistake!
Heard that we have a patch of their own pyqt5.4 ?
 Where should I go to find him?
Or how can I solve this problem?

Ghostkeeper, 2016-08-22:
Yes, we found a deadlock bug in Qt during Cura's development. When we submitted this bug to Qt's developers, they came with a patch. We apply that patch in our build system, and PyQt incorporated the patch in version 5.6. The patch files are [here](https://github.com/Ultimaker/cura-build/blob/master/qpyqmlobject.cpp.patch) and [here](https://github.com/Ultimaker/cura-build/blob/master/qpyqmlobject.h.patch).

As a consequence, you'll need one of these three versions of Qt to run Cura:
* Patched PyQt 5.4.
* Patched PyQt 5.5.
* PyQt 5.6.

fieldOfView, 2016-08-22:
I have edited the title of your issue to reflect that you are in fact not using 2.1 but the soon-to-be 2.2 from the master branch

twosilly-2B, 2016-08-23:
I'm sorry, please forgive me for my stupidity!
I tried to install pyqt5.4 on the base of my python3.4, but I failed!
How am I going to run this pyqt5 patch alone or how to install pyqt5.6 in the windows7-64 bit, python3.4?
A low level problem from a student!
@fieldOfView 
@Ghostkeeper

fieldOfView, 2016-08-23:
I have not tried this myself recently (as I don't want to mess up my development machine at this point), but have you tried ```pip3 install PyQt5``` (as per https://www.riverbankcomputing.com/software/pyqt/download5)?

twosilly-2B, 2016-08-23:
Yeah, I tried.“ pip install PyQt5”

Maybe it was because I was windows's sake. He was in the wrong!
Collecting PyQt5
  Could not find a version that satisfies the requirement PyQt5 (from versions:
)
No matching distribution found for PyQt5
I also tried to go directly to the pyqt's official website to download and install it!
 But I found that pyqt5.6 only python3.5 installation package!
 Then I tried to run it directly with the code, the results I failed again! @fieldOfView

fieldOfView, 2016-08-23:
Ah, yes. I'll check my patched binaries for python 3.4 and will post them somewhere for now.

twosilly-2B, 2016-08-23:
I tried to replace!
E:\cura\cura-2.1\cura-2.1>python cura_app.py
Traceback (most recent call last):
  File "cura_app.py", line 10, in <module>
    from UM.Platform import Platform
  File "E:\cura\cura-2.1\cura-2.1\UM\__init__.py", line 9, in <module>
    from .Application import Application
  File "E:\cura\cura-2.1\cura-2.1\UM\Application.py", line 17, in <module>
    from UM.Preferences import Preferences
  File "E:\cura\cura-2.1\cura-2.1\UM\Preferences.py", line 8, in <module>
    from UM.MimeTypeDatabase import MimeTypeDatabase, MimeType #To register the
MIME type of the preference file.
  File "E:\cura\cura-2.1\cura-2.1\UM\MimeTypeDatabase.py", line 4, in <module>
    from PyQt5.QtCore import QMimeDatabase
ImportError: DLL load failed: Cannot find the specified program。

why？
@fieldOfView Please.

fieldOfView, 2016-08-23:
I don't know why...

As a general note it seems that you have combined the Cura and the Uranium repository into a single folder. This is not recommended. We recommend to keep them in separate folders and add the Uranium folder to your PYTHONPATH.

awhiemstra, 2016-09-05:
Note that the PyQt5 wheels are only available for Python 3.5. Our builds currently use Python 3.4 and a patched PyQt 5.5.1.

twosilly-2B, 2016-09-10:
I solve!I made a python3.4 version of the pyqt5.4 no bug version of exe! X64! I hope to be useful to you!

twosilly-2B, 2016-09-10:
I wanted to put it straight up! Suddenly I didn't know how to do it!
There is a need to contact me directly, TwoSilly@foxmail.com

Ghostkeeper, 2016-09-11:
Normally, a link to some third-party file hosting service would suffice. Try [Mega](https://mega.nz/) or somesuch, or provide instructions on how to reproduce it.

closed 2016-09-17 by thopiekar