Ultimaker/Cura #939: Ubuntu 16.04 LTS Upgrade Issue

opensourcefan, 2016-08-22:
Hello, I just upgraded to 16.04LTS over 14.04LTS. I had Cura running fine yesterday but today after the upgrade it won't start. It appears that my Python version has changed from 3.4 to 3.5.2 and I think this may be the issue. Here is my terminal output:

xxxxxxxx@ubuntu:~$ cura
/usr/bin/cura: line 3: python3.4: command not found
xxxxxxxx@ubuntu:~$ python3 -V
Python 3.5.2

I did try to change the 3.4 to 3.5 and 3.5.2 but I just get a segmentation fault warning afterwords. 

2.1.3
Cura Log has no information for today.

Ghostkeeper, 2016-08-22:
We've seen it too on our test machine. Thanks for the report!

I think it's still unsolved in master, but @awhiemstra has been on it. Our internal issue (CURA-1762) is assigned at least. We aim to get this one for 2.2.

opensourcefan, 2016-08-25:
I just found this, I should have looked sooner I think ;). Upon the FAQ's recommendation I installed the cura PPA to get the latest builds. Installed [sudo apt-get install cura] from terminal but it didn't work for some reason until I "upgraded" [sudo apt-get upgrade cura]. It works now after that. Upgrading downgraded my Python from 3.5.2 to 2.7.12. So far so good. Looking forward to 2.2

https://ultimaker.com/en/resources/21362-cura-2-1-linux-not-working

thopiekar, 2016-08-29:
The reason is that the package provided by Ultimaker uses an higher version than used in my PPA.
So before installing, it is needed to remove the cura package first and than you can easily reinstall cura from my PPA.

opensourcefan, 2016-08-29:
Thank you. I removed it via Ubuntu Software but it didn't remove completely. Maybe I should have removed via command line instead.

thopiekar, 2016-09-02:
No problem :wink: 
To make sure everything got removed you'll need to run `sudo apt-get remove --purge cura`

Let me know if there are further problems.

closed 2016-09-02 by thopiekar

Whytehorse, 2017-01-06:
I just installed the official binary from cura. I got
```
 $ cura
/usr/bin/cura: line 3: python3.4: command not found
```
This is a bug in cura. No idea how to report it or how to fix it. Where's the ppa instructions mentioned in the previous replies?

shadybug, 2017-01-06:
The PPA can be found here: https://launchpad.net/~thopiekar/+archive/ubuntu/cura
so, in short, here's the steps you want to take:
`sudo apt-get remove --purge cura
sudo add-apt-repository ppa:thopiekar/cura
sudo apt-get update
sudo apt-get install cura`
I was getting the exact same error and this fixed it for me!

Ghostkeeper, 2017-01-06:
That was a bug in Cura 2.3 because it required Python 3.4 specifically. We solved this for Cura 2.4.

Whytehorse, 2017-01-06:
Thank you! Can someone let the cura website maintainer know they're distributing 2.3 on the official website?

Whytehorse, 2017-01-06:
I got it to work! It requires sudo to see the USB port. Is this necessary or can I add my user to some group or???

thopiekar, 2017-01-06:
Ouch! Of course it needs. Most distributions don't set the membership be default. So you have to join `dialout`, `modem` or however they are called.

The easiest way to find out which group membership is needed, is to `ls -la /dev/tty*` and look which group takes your `ttyACM0` or `ttyUSB0`.

However, likely you'll need to repair permissions of your configuration files now.

thopiekar, 2017-01-06:
@Whytehorse 2.4 is currently in beta, that's why it is more or less hidden on the website.

teclauin, 2017-09-29:
EN LAS ULTIMAS VERSIONES DEL CURA NO RECONOCE LA IMPRESORA ANET A8 EN UBUNTU

Ghostkeeper, 2017-10-05:
Anet A8 was never added, not on Ubuntu or otherwise.

teclauin, 2017-10-06:
anet a8 usa marlin es la misma iprusa i3