Ultimaker/Cura #942: Enhancement: Use Raft and Skirt at same time.

opensourcefan, 2016-08-23:
I generally use the Skirt to clean my nozzle and get everything flowing before the actual print starts. I use the Raft for adhesion. It seems as though I can only use one or the other as they're both in the same drop down menu. Would it be possible to separate the Skirt from Raft and Brim? Or am I missing something obvious, that does happen often ;)

fieldOfView, 2016-08-23:
We feel the same way.

jweather, 2016-12-09:
I would like to be able to use both.  One edge of my raft frequently peels up because the filament hasn't made it out of the nozzle yet.  I use the skirt to make sure the filament is flowing smoothly first.

cinderblock, 2016-12-15:
I'm having similar issues with my Raft peeling up. I think this would be fixed by better purge options or simply printing a skirt or brim to my base layer.

Honestly, I rather like the skirt as a visual aid when starting the print.

I think the options for Brim, Raft, and Skirt should be split as the OP originally suggested.

If users find that using a Raft does the purge they need, great; they don't need to turn it on.

Ghostkeeper, 2017-05-01:
This issue was removed from our planning because it had been inactive for three months.

jgabriels, 2017-05-04:
I also would like a skirt plus raft option.  Either that or a Two or three inch lead in strip before starting the raft.  It seems my printer always misses the first couple inches of the raft.

3DPrintedWaffles, 2018-02-24:
+1 for being able to do both.  Brim would come out much better/reliably if primed with a skirt first.

closed 2018-11-26 by Appesteijn

cinderblock, 2018-11-26:
Why was this closed?

Appesteijn, 2018-11-26:
1. To make clear we (the Cura team) is not going to do this and don't give any false hopes that we will do this in some distant future. Of course someone else can implement this, but we don't have the time.
2. To clean up the issue list so we can focus on the things that are critical and need to be fixed.

3DPrintedWaffles, 2018-11-26:
Is it so complicated to add that it isn't worth delivering higher quality, more reliable prints with your product?  Seems like it would be a slam dunk.  

I'm no programmer but it seems as simple as (Do A), then (Do B) then continue as normal.

qubitnerd, 2019-02-24:
this would indeed be a great feature , however ....
Since the problem appears to be the raft/brim doesnt stick to the bed especially the initial lines put down since the nozzle hasnt been primed with filament yet , one solution would be for you to add a custom gcode before the print , a gcode that prints a line on the edge of the bed (or a square all around the bed) before heading to print your model , this way the nozzle is well primed as it prints the raft/brim

LeosWelt, 2020-03-15:
I am not only using Cura for slicing, and every software has its pros and cons. Cura does the best job with almost every 3D-model i have created until now...
But the buidplate adhesion, other softwares have other techniques. KISSlicer has the worst way to parse the 3D-data, seems to calculate with binary floatingpoint, but it has the ability to put a skirt around the raft and a second skirt above each raft.
Prusa Slicer has some problems with thin surfaces, creating interesting printouts with artifacts. But the Brim with a raft on top, that is interesting, creating a build-thing your prints fix to, even when printing thin wings for RC-planes.
Would be nice if skirt-raft-brim-things get updated or be configurable by an addon for people printing thin things and need a good adhesion even wehn the thing is 300mm in height...

jdobyns, 2022-01-06:
Cura, this ticket solidified in my mind just how irrelevant your product is.