Ultimaker/Cura #946: Bed Temperature not saving on New Profile Creation

bionicbone, 2016-08-25:
I'm using v2.1.3

Set the profile in the main screen for PLA with a Bed Temperature of 73c (anything other than 60c)

Drop down the profile Combo and select "add profile...", note the current settings in this screen have defaulted the Bed Temperature back to 60c.

OK, so now try again, this time select "Manage Profiles...", this time the current settings temperature is correct at 73c. However, Add the profile and the bed temperature changes back to 60c and its in the profile as 60c.

Basically, the bed has an obsession about being 60c :)

I'd be grateful if you could fix this because I keep printing and 60c is too cold to hold the parts.

Cheers
Kev

bionicbone, 2016-08-25:
This happens with Retraction Distance also :(

Ghostkeeper, 2016-08-26:
Known issue. All the settings in the material profiles were overwriting the user-chosen settings. This has been fixed in Cura 2.3. There'll be a public beta next week if all goes well.

As a workaround, I made an empty material profile for you. Put this .cfg file in your profile folder: [generic.cfg.zip](https://github.com/Ultimaker/Cura/files/438773/generic.cfg.zip)
Your profile folder is in C:/Users/<bionicbone>/AppData/cura/profiles (for Windows). If you put it there and then start Cura, it should have a new material "Generic" which doesn't set the bed temperature to 60 or the retraction distance to 6.5. You would then have to set these values yourself, but at least you can set them yourself.

closed 2016-08-26 by Ghostkeeper