Ultimaker/Cura #947: [master-e6aec0c6] profile dropdown doesn't update properly for a dual extrusion printer

markwal, 2016-08-25:
Steps:
- Add and select Ultimaker Original Dual Extrusion Printer
- Advanced Print Setup
- Change something (like layer height)
- Save as new profile
- Choose that profile

Result:
Note that the new profile seems to take effect and is checked in the dropdown, but the profile control stubbornly says "Normal Quality" regardless of what profile is selected.

Expected:
Same as in single extrusion, profile box shows currently selected profile.

I know I'm ahead of things since no official release contains dual support, just thought you might appreciate a bug report.  And this doesn't affect the functionality much so I don't need a fix or anything, but y'all might want to fix it before release.

sedwards2009, 2016-08-25:
Thanks for the report. The profiles have been getting quite a work out this week going between our programmers and QA people and back. We've got a couple important profile related issues at the top of our internal bug tracker at the moment. So we should be able to get this sorted out soon.

markwal, 2017-06-28:
I don't think this is a problem in 2.5+ any more

closed 2017-06-28 by markwal